riscv-collab/riscv-gnu-toolchain #615: hello_world.c:1:10: fatal error: stdio.c: No such file or directory

Kmambu, 2020-03-31:
Hi everyone,
I tried to build myself a cross-compilation toolchain using this repository and I cannot manage to pass a simple «Hello World» test :-/

```c
#include <stdio.c>                                                               
#include <stdlib.c>                                                              

int main(void) {                                                                 
    print("Hello World\n");                                                      
    return EXIT_SUCCESS;                                                         
}                                                                                         
```

Here is the `configure` and `make` commands I used for this : 
```bash
./configure --prefix=/home/kevin/Documents/Doctorat/riscv --with-arch=rv32gc  --with-abi=ilp32d
make -j5 linux
```

These are te searched paths I can see using the `-v` flag : 
```
#include "..." search starts here:
#include <...> search starts here:
 /home/kevin/Documents/Doctorat/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/../../../../riscv32-unknown-linux-gnu/include/c++/9.2.0
 /home/kevin/Documents/Doctorat/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/../../../../riscv32-unknown-linux-gnu/include/c++/9.2.0/riscv32-unknown-linux-gnu
 /home/kevin/Documents/Doctorat/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/../../../../riscv32-unknown-linux-gnu/include/c++/9.2.0/backward
 /home/kevin/Documents/Doctorat/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/include
 /home/kevin/Documents/Doctorat/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/include-fixed
 /home/kevin/Documents/Doctorat/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/../../../../riscv32-unknown-linux-gnu/include
 /home/kevin/Documents/Doctorat/riscv/sysroot/usr/include
```

My OS environment is ArchLinux with Kernel Version 5.5.9. I did not see any issues similar to this one on the board and I followed the README, I believe, accurately. Am I missing something ?

kito-cheng, 2020-03-31:
Maybe stduio.h and stdlib.h are what you need ?
```
#include <stdio.h>
#include <stdlib.h>
```

Kmambu, 2020-03-31:
OMG, I'm so ashamed XD
Thanks, man !

closed 2020-03-31 by Kmambu

aswaterman, 2020-03-31:
For future reference, it's always helpful to try compiling the program for your host machine before trying to dig into the RISC-V stuff :)