riscv-collab/riscv-gnu-toolchain #617: make linux error: The riscv64 is not supported.

unicornx, 2020-04-09:
I see a similar issue https://github.com/riscv/riscv-gnu-toolchain/issues/604

I build the Newlib cross-compiler successfully also, but failed make linux.
Against 604, my failure occurred in another phase, following is snapshot I got for your reference:

```
mkdir -p stamps/ && touch stamps/build-gcc-linux-stage1
rm -rf stamps/build-glibc-linux-rv64imafdc-lp64d build-glibc-linux-rv64imafdc-lp64d
mkdir build-glibc-linux-rv64imafdc-lp64d
cd build-glibc-linux-rv64imafdc-lp64d && \
	CC="riscv64-unknown-linux-gnu-gcc " \
	CXX="riscv64-unknown-linux-gnu-g++ " \
	CFLAGS="  -mcmodel=medlow -g -O2 " \
	CXXFLAGS="  -mcmodel=medlow -g -O2 " \
	ASFLAGS=" -mcmodel=medlow " \
	/home/u/ws/riscv64-linux/riscv-gnu-toolchain/riscv-glibc/configure \
	--host=riscv64-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/home/u/ws/riscv64-linux/riscv-gnu-toolchain/linux-headers/include \
	--disable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv64-unknown-linux-gnu-gcc  accepts -g... yes
checking for gcc... gcc
checking for riscv64-unknown-linux-gnu-readelf... riscv64-unknown-linux-gnu-readelf
checking whether we are using the GNU C++ compiler... no
checking whether riscv64-unknown-linux-gnu-g++  accepts -g... no
checking whether riscv64-unknown-linux-gnu-g++  can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arm hppa i386 m68k microblaze mips nios2 powerpc s390 sh sparc tile x86_64 
checking for use of fpu sysdeps directories... yes
checking for -fstack-protector... yes
checking for -fstack-protector-strong... yes
checking for -fstack-protector-all... yes
checking for assembler and linker STT_GNU_IFUNC support... no
checking for gcc attribute ifunc support... no
checking if compiler warns about alias for function with incompatible types... yes
configure: WARNING: gcc emits a warning for alias between functions of incompatible types
configure: WARNING: Multi-arch is disabled.
checking sysdep dirs... configure: error: The riscv64 is not supported.
Makefile:234: recipe for target 'stamps/build-glibc-linux-rv64imafdc-lp64d' failed
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 1
```
I am working on Ubuntu 18.04
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic
```

jim-wilson, 2020-04-09:
Master branch should work.

This line is wrong
checking for sysdeps preconfigure fragments... aarch64 alpha arm hppa i386 m68k microblaze mips nios2 powerpc s390 sh sparc tile x86_64 

The configure script does "ls sysdeps/*/preconfigure" and then runs it through sed to extract the isa dir names.  cky and riscv are missing from the list, and tile is present but shouldn't be.  It doesn't seem like that could be a ls or sed or shell problem.  Looking at glibc history, I see that tile support was removed in December 2017.  So it looks like you have an old copy of glibc, maybe before the riscv support was added?  I don't know how you managed to do that.

What does "git status" report in the riscv-gnu-toolchain top level dir, and in the riscv-glibc dir?  With "git log -1" in riscv-glibc, you should see this

commit 06983fe52cfe8e4779035c27e8cc5d2caab31531
Author: Alistair Francis <alistair.francis@wdc.com>
Date:   Fri Feb 1 15:14:33 2019 +0800

    To avoid an out of bound array check when building with GCC 8.2 make the array one size larger.
    
    Signed-off-by: Alistair Francis <alistair.francis@wdc.com>

unicornx, 2020-04-10:
Thank you, Jim.

I checked repos on **github**, below are the default branch for following submodules:
- https://github.com/riscv/riscv-glibc --> riscv-glibc-2.26
- https://github.com/riscv/riscv-gcc -->  riscv-gcc-8.3.0
- https://github.com/riscv/riscv-binutils-gdb --> riscv-binutils-2.31.1
- https://github.com/riscv/riscv-dejagnu --> riscv-dejagnu-1.6
- https://github.com/riscv/riscv-newlib --> riscv-newlib-3.1.0

I am using **master** for my glibc, and it really looks a bit old (still live in 2017 ...), and I believe that's make my building failed. Then I changed branches same as upon and now I can make linux.

btw, I have a minor question:
as README tells, we should use following commands:
```
$ git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
```
I see this will download lots of sourcecodes recursively, especially for **qemu**, this will enter very deep directories and clone many many sub modules.  So my question is: for riscv-toolchain build, do we have dependency on these more components? I believe we just need clone the first level for qemu repo, and that's what I have done and seems it works (no failure when building at least till now).

For toolchain's other first-level submodules  (i.e. riscv-glibc/riscv-gcc/riscv-binutils-gdb/riscv-dejagnu/riscv-newlib), no such issue due to they don't have submodules.

jim-wilson, 2020-04-10:
The master branch for riscv-glibc is wrong.  You want the master branch for riscv-gnu-toolchain, and then update submodules and that will give the right version of glibc and all of the other submodules.  So master branch riscv-gnu-toolchain will currently pull the branch riscv-glibc-2.29 for submodule riscv-glibc, but in the future that will eventually be updated to a newer glibc version.  If you just use master of riscv-gnu-toolchain then you will always have the latest version of everything.  Don't try updating submodules individually unless you know what you are doing.

qemu is only used for toolchain testing, and isn't built by default.  It is useful if you want to run the toolchain testsuites.  And it is useful if you want to try compiling programs and running them in a simulator.  If you do build qemu, you will probably need more than the first level of it.  You probably need all of it.

jim-wilson, 2020-04-10:
I don't have owner permissions for any of the submodules, so I can't change their default branches.  Only the default branch of riscv-gnu-toolchain is useful, because we are maintaining master as top of tree.  For submodules, each upstream release is on its own branch.

unicornx, 2020-04-13:
Thank you Jim, I have no more questions. I will close this issue.

closed 2020-04-13 by unicornx