riscv-collab/riscv-gnu-toolchain #618: rv32i: Weird optimizer behaviour with -Os

nshcat, 2020-04-09:
Hello,
I'm seeing weird optimizer behavior with the current master, where code with optimizations disabled is often significantly smaller than code compiled with `-Os`.

I have a small example here:

**Optimizations disabled:**
```
    GPIO_OUT &= 0xF0;
 444:	000047b7          	lui	a5,0x4
 448:	03178793          	addi	a5,a5,49 # 4031
 44c:	0007a703          	lw	a4,0(a5)
 450:	000047b7          	lui	a5,0x4
 454:	03178793          	addi	a5,a5,49 # 4031
 458:	0f077713          	andi	a4,a4,240
 45c:	00e7a023          	sw	a4,0(a5)
```

**Optimize for size (`-Os`):**
```
   GPIO_OUT &= 0xF0;
 55c:	0314c783          	lbu	a5,49(s1)
 560:	00144703          	lbu	a4,1(s0)
 564:	00244703          	lbu	a4,2(s0)
 568:	00344703          	lbu	a4,3(s0)
 56c:	0f07f793          	andi	a5,a5,240
 570:	0314c703          	lbu	a4,49(s1)
 574:	02f488a3          	sb	a5,49(s1)
 578:	00144703          	lbu	a4,1(s0)
 57c:	000400a3          	sb	zero,1(s0)
 580:	00244703          	lbu	a4,2(s0)
 584:	0187d793          	srli	a5,a5,0x18
 588:	00040123          	sb	zero,2(s0)
 58c:	00344703          	lbu	a4,3(s0)
 590:	00f401a3          	sb	a5,3(s0)
```

It seems like the compiler is trying to be smart with byte accesses, but ends up blowing up the code size. In total, this causes an increase of over 50% of the firmware binary in my projects. Is this expected behaviour?

kito-cheng, 2020-04-09:
Did you mind provide minimal test case to reproduce that? it seems like GPIO_OUT is unaligned address, but need confirmed by source code.

nshcat, 2020-04-09:
Thank you, misaligned I/O read was indeed the issue.

closed 2020-04-09 by nshcat