riscv-collab/riscv-gnu-toolchain #620: unable to clone the repo in windows cygwin

quantrpeter, 2020-04-14:
hi, 

```
Submodule path 'qemu/capstone': checked out '22ead3e0bfdb87516656453336160e0a37b066bf'
Submodule path 'qemu/dtc': checked out '88f18909db731a627456f26d779445f84e449536'
Submodule path 'qemu/roms/QemuMacDrivers': checked out '90c488d5f4a407342247b9ea869df1c2d9c8e266'
Submodule path 'qemu/roms/SLOF': checked out 'a5b428e1c1eae703bdd62a3f527223c291ee3fdc'
Submodule path 'qemu/roms/edk2': checked out '20d2e5a125e34fc8501026613a71549b2a1a3e54'
Submodule 'SoftFloat' (https://github.com/ucb-bar/berkeley-softfloat-3.git) registered for path 'qemu/roms/edk2/ArmPkg/Library/ArmSoftFloatLib/berkeley-sof
tfloat-3'
Submodule 'CryptoPkg/Library/OpensslLib/openssl' (https://github.com/openssl/openssl) registered for path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/open
ssl'
Cloning into 'C:/cygwin64/home/Peter/workspace/riscv-gnu-toolchain/qemu/roms/edk2/ArmPkg/Library/ArmSoftFloatLib/berkeley-softfloat-3'...
Cloning into 'C:/cygwin64/home/Peter/workspace/riscv-gnu-toolchain/qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl'...
Submodule path 'qemu/roms/edk2/ArmPkg/Library/ArmSoftFloatLib/berkeley-softfloat-3': checked out 'b64af41c3276f97f0e181920400ee056b9c88037'
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl': checked out '50eaac9f3337667259de725451f201e784599687'
Submodule 'boringssl' (https://boringssl.googlesource.com/boringssl) registered for path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/boringssl'
Submodule 'krb5' (https://github.com/krb5/krb5) registered for path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/krb5'
Submodule 'pyca.cryptography' (https://github.com/pyca/cryptography.git) registered for path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/pyca-cryp
tography'
Cloning into 'C:/cygwin64/home/Peter/workspace/riscv-gnu-toolchain/qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/boringssl'...
Cloning into 'C:/cygwin64/home/Peter/workspace/riscv-gnu-toolchain/qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/krb5'...
Cloning into 'C:/cygwin64/home/Peter/workspace/riscv-gnu-toolchain/qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/pyca-cryptography'...
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/boringssl': checked out '2070f8ad9151dc8f3a73bffaa146b5e6937a583f'
Submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/krb5': checked out 'b9ad6c49505c96a088326b62a52568e3484f2168'
error: unable to create file vectors/cryptography_vectors/x509/PKITS_data/certpairs/SeparateCertificateandCRLKeysCA2CertificateSigningCACertforwardcrossCer
ificatePair.cp: Filename too long
error: unable to create file vectors/cryptography_vectors/x509/PKITS_data/certpairs/SeparateCertificateandCRLKeysCA2CertificateSigningCACertreversecrossCer
ificatePair.cp: Filename too long
error: unable to create file vectors/cryptography_vectors/x509/PKITS_data/certpairs/SeparateCertificateandCRLKeysCertificateSigningCACertforwardcrossCertif
icatePair.cp: Filename too long
error: unable to create file vectors/cryptography_vectors/x509/PKITS_data/certpairs/SeparateCertificateandCRLKeysCertificateSigningCACertreversecrossCertif
icatePair.cp: Filename too long
Unable to checkout '09403100de2f6f1cdd0d484dcb8e620f1c335c8f' in submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl/pyca-cryptography'
Failed to recurse into submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl'
Submodule path 'qemu/roms/ipxe': checked out 'de4565cbe76ea9f7913a01f331be3ee901bb6e17'
Submodule path 'qemu/roms/openbios': checked out 'c79e0ecb84f4f1ee3f73f521622e264edd1bf174'
Submodule path 'qemu/roms/openhackware': checked out 'c559da7c8eec5e45ef1f67978827af6f0b9546f5'
Submodule path 'qemu/roms/qemu-palcode': checked out 'bf0e13698872450164fa7040da36a95d2d4b326f'
Submodule path 'qemu/roms/seabios': checked out 'a5cab58e9a3fb6e168aba919c5669bea406573b4'
Submodule path 'qemu/roms/seabios-hppa': checked out '0f4fe84658165e96ce35870fd19fc634e182e77b'
Submodule path 'qemu/roms/sgabios': checked out 'cbaee52287e5f32373181cff50a00b6c4ac9015a'
Submodule path 'qemu/roms/skiboot': checked out '261ca8e779e5138869a45f174caa49be6a274501'
Submodule path 'qemu/roms/u-boot': checked out 'd3689267f92c5956e09cc7d1baa4700141662bff'
Submodule path 'qemu/roms/u-boot-sam460ex': checked out '60b3916f33e617a815973c5a6df77055b2e3a588'
Submodule path 'qemu/slirp': checked out 'f0da6726207b740f6101028b2992f918477a4b08'
Submodule path 'qemu/tests/fp/berkeley-softfloat-3': checked out 'b64af41c3276f97f0e181920400ee056b9c88037'
Submodule path 'qemu/tests/fp/berkeley-testfloat-3': checked out '5a59dcec19327396a011a17fd924aed4fec416b3'
Submodule path 'qemu/ui/keycodemapdb': checked out '6b3d716e2b6472eb7189d3220552280ef3d832ce'
Failed to recurse into submodule path 'qemu/roms/edk2'
Submodule path 'riscv-binutils': checked out 'd7f734bc7e9e5fb6c33b433973b57e1eed3a7e9f'
Submodule path 'riscv-dejagnu': checked out '4ea498a8e1fafeb568530d84db1880066478c86b'
Submodule path 'riscv-gcc': checked out '54945eb8ad5a066da2d4e6a62ffeb513d341eb41'
Submodule path 'riscv-gdb': checked out 'c3eb4078520dad8234ffd7fbf893ac0da23ad3c8'
Submodule path 'riscv-glibc': checked out '06983fe52cfe8e4779035c27e8cc5d2caab31531'
Submodule path 'riscv-newlib': checked out 'f289cef6be67da67b2d97a47d6576fa7e6b4c858'
Failed to recurse into submodule path 'qemu'
/home/Peter/workspace/riscv-gnu-toolchain>
```

colinschmidt, 2020-04-14:
There appear to be several solutions for this "filename too long" problem in this stack overflow post:
https://stackoverflow.com/questions/21123415/git-pull-aborted-with-error-filename-too-long

joshua23333, 2020-10-07:
同问这个问题，感觉这个不是filename too long的问题吧，咋感觉这是上不了外网的问题呢，下面这句话：
Failed to recurse into submodule path 'qemu/roms/edk2/CryptoPkg/Library/OpensslLib/openssl'

kito-cheng, 2020-10-07:
@joshua23333 maybe you could try `git clone https://github.com/openssl/openssl` to check is the network issue or not?

TommyMurphyTM1234, 2022-09-27:
No update/activity in several years and the issue mentioned is nothing to do with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234