riscv-collab/riscv-gnu-toolchain #622: Bare minimum -march and -mabi support for linux cross compiler? Build fails with 32i as based.

shivampotdar, 2020-04-19:
I tried compiling the newlib compiler for almost all possible targets using multilib-generator format similar to that in sifive freedom tools and that works for all -march and -mabi the combinations including rv32i and rv32e:
`./multilib-generator rv32e-ilp32e--c rv32ea-ilp32e--m rv32em-ilp32e--c rv32eac-ilp32e-- rv32emac-ilp32e-- rv32i-ilp32--c rv32ia-ilp32--m rv32im-ilp32--c rv32if-ilp32f-rv32ifd-c rv32iaf-ilp32f-rv32imaf,rv32iafc-d rv32imf-ilp32f-rv32imfd-c rv32iac-ilp32-- rv32imac-ilp32-- rv32imafc-ilp32f-rv32imafdc- rv32ifd-ilp32d--c rv32imfd-ilp32d--c rv32iafd-ilp32d-rv32imafd,rv32iafdc- rv32imafdc-ilp32d-- rv64i-lp64--c rv64ia-lp64--m rv64im-lp64--c rv64if-lp64f-rv64ifd-c rv64iaf-lp64f-rv64imaf,rv64iafc-d rv64imf-lp64f-rv64imfd-c rv64iac-lp64-- rv64imac-lp64-- rv64imafc-lp64f-rv64imafdc- rv64ifd-lp64d--m,c rv64iafd-lp64d-rv64imafd,rv64iafdc- rv64imafdc-lp64d--
`
For Linux, I tried with removing all the e based targets:
`./multilib-generator rv32i-ilp32--c rv32ia-ilp32--m rv32im-ilp32--c rv32if-ilp32f-rv32ifd-c rv32iaf-ilp32f-rv32imaf,rv32iafc-d rv32imf-ilp32f-rv32imfd-c rv32iac-ilp32-- rv32imac-ilp32-- rv32imafc-ilp32f-rv32imafdc- rv32ifd-ilp32d--c rv32imfd-ilp32d--c rv32iafd-ilp32d-rv32imafd,rv32iafdc- rv32imafdc-ilp32d-- rv64i-lp64--c rv64ia-lp64--m rv64im-lp64--c rv64if-lp64f-rv64ifd-c rv64iaf-lp64f-rv64imaf,rv64iafc-d rv64imf-lp64f-rv64imfd-c rv64iac-lp64-- rv64imac-lp64-- rv64imafc-lp64f-rv64imafdc- rv64ifd-lp64d--m,c rv64iafd-lp64d-rv64imafd,rv64iafdc- rv64imafdc-lp64d--
`
But still after running for some time the build process fails with:
```
In file included from /sysroot/usr/include/pthread.h:27,
                 from ./gthr-default.h:35,
                 from ../../../../.././riscv-gcc/libgcc/gthr.h:148,
                 from ../../../../.././riscv-gcc/libgcc/libgcov-interface.c:27:
/sysroot/usr/include/bits/setjmp.h:35:3: error: #error unsupported FLEN
   35 | # error unsupported FLEN
      |   ^~~~~

```
Can someone please guide on what is going wrong. I could not find any resources on what shoiuld be the minimal -march and -mabi for the linux compiler, although by default the toolchain provides rv32/64ima. I want to attempt compiling for i/im based targets.

jim-wilson, 2020-04-19:
Linux has much higher support requirements than embedded.  Any reasonable march/mabi should work embedded, but for linux only a small number of combinations will work.

We don't have a convenient list of linux supported arch/abi combinations.  I can say that A (atomic) is required, we support both float+double and soft-float, and we support rv32 and rv64.  So I would expect rv32imac/ilp32, rv32gc/ilp32, rv32gc/ilp32d, rv64imac/lp64, rv64gc/lp64, and rv64gc/lp64d to work.  Those are the ones that the regression/Makefile builds.  I don't know about anything else.  If you want a ful list, you will have to experiment to find what works and doesn't.  Some combinations might build and fail at runtime, so there is a nontrivial amount of work required to get the list.

shivampotdar, 2020-04-20:
Thanks a lot @jim-wilson! Y

closed 2020-04-20 by shivampotdar

fanghuaqi, 2021-01-06:
Hi @jim-wilson , Does currently `rv64imafc + lp64f` supported in Linux glibc toolchain now?

jim-wilson, 2021-01-06:
In theory single-float only should work.  I don't know if anyone is testing the linux kernel or glibc support.  I think only the soft-float and double-float (f+d) are being tested in practice.

jim-wilson, 2021-01-06:
Actually for single-float support, see riscv-glibc issue #52.  It was apparently removed due to lack of time when the code was upstreamed, and never readded due to lack of volunteers.

fanghuaqi, 2021-01-07:
Hi @jim-wilson , thank you for your explaination, we will just focus on rv64gc, when time is free, we can look into glibc + f support.