riscv-collab/riscv-gnu-toolchain #623: Trouble with *this* pointer (riscv32-unknown-elf-g++)

nbelanger99, 2020-04-24:
Hi, please reference 
https://github.com/ros-drivers/rosserial/issues/496
as I have a feeling this could be an issue with the compiler itself.

kito-cheng, 2020-04-27:
Could you try compile with `-fno-delete-null-pointer-checks` to see what happen? I not sure how to reproduce the problem, so I only can guess you might hit the null pointer check optimization which introduced from GCC 6.

https://gcc.gnu.org/gcc-6/changes.html
- Value range propagation now assumes that the this pointer in C++ member functions is non-null. This eliminates common null pointer checks but also breaks some non-conforming code-bases (such as Qt-5, Chromium, KDevelop). As a temporary work-around -fno-delete-null-pointer-checks can be used. Wrong code can be identified by using -fsanitize=undefined.

kito-cheng, 2020-10-12:
Close this issue, long time no further discussion/response, feel free to create issue or reopen this issue if you have further question.

closed 2020-10-12 by kito-cheng