riscv-collab/riscv-gnu-toolchain #627: Purpose of riscv-gdb and riscv-binutils

bluewww, 2020-04-29:
Why is `riscv-binutils-gdb` split into `riscv-gdb` and `riscv-binutils`? As far as I can see its nearly identical code but from two different branches.

kito-cheng, 2020-04-30:
We intend to maintain those two repo based on stable release plus few serious bug fix, however GDB and binutils are shared same source tree on upstream/FSF, but they have different release scheduling, so that's why they using similar code but different branch :)

Last 3 stable release date for GDB and binutils:
|                      | GDB  | binutils |
| ---------------- | ---------| --------- |
| Version / Release date | 9.1 / 2020-02-08 |  2.34 / 2020-02-01 |
|                                      | 8.3 / 2019-05-11 | 2.33.1 / 2019-10-12 |
|                                      | 8.2 / 2018-09-05 | 2.32 / 2019-02-02 |

bluewww, 2020-05-01:
So when compiling `riscv-gnu-toolchain` and running `make install` then `gdb` will be copied from `riscv-gdb` and `binutils` from `riscv-binutils` or is everything taken from one subrepo?

Nelson1225, 2020-05-04:
Yes, `binutils` and `gdb` are taken from separate subrepo.  The former is taken from `riscv-binutils`, and the later is taken from `riscv-gdb`, just like you said.

bluewww, 2020-05-04:
Thanks for the help

closed 2020-05-04 by bluewww