riscv-collab/riscv-gnu-toolchain #632: fail to make Linux cross-compiler

jasonlee1001, 2020-05-08:
Hi all,
I want to generate elf-gcc and linux-gcc to use.
elf-gcc (Newlib) is done with no errors.
But fail to build "Installation (Linux)" and "Installation (Linux multilib)" with the same error messages.

 $ ./configure --prefix=/opt/riscv
 $ make linux
 After very long compiling time, I got message below. Can anyone know how to fix it ?
 Thank you.
 My environment: Ubuntu 64, GCC version: 4.8.3 

g++    -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o f951 \
	fortran/arith.o fortran/array.o fortran/bbt.o fortran/check.o fortran/class.o fortran/constructor.o fortran/cpp.o fortran/data.o fortran/decl.o fortran/dump-parse-tree.o fortran/error.o fortran/expr.o fortran/interface.o fortran/intrinsic.o fortran/io.o fortran/iresolve.o fortran/match.o fortran/matchexp.o fortran/misc.o fortran/module.o fortran/openmp.o fortran/options.o fortran/parse.o fortran/primary.o fortran/resolve.o fortran/scanner.o fortran/simplify.o fortran/st.o fortran/symbol.o fortran/target-memory.o  fortran/convert.o fortran/dependency.o fortran/f95-lang.o fortran/trans.o fortran/trans-array.o fortran/trans-common.o fortran/trans-const.o fortran/trans-decl.o fortran/trans-expr.o fortran/trans-intrinsic.o fortran/trans-io.o fortran/trans-openmp.o fortran/trans-stmt.o fortran/trans-types.o fortran/frontend-passes.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a -lz libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  attribs.o \
	 -lmpc -lmpfr -lgmp -rdynamic -ldl  -lz
fortran/arith.o: In function `gfc_mpfr_to_mpz(__mpz_struct*, __mpfr_struct*, locus*)':
/home/jason/work/riscv-gnu-toolchain/build-gcc-linux-stage2/gcc/../.././riscv-gcc/gcc/fortran/arith.c:51: undefined reference to `mpfr_get_z_exp'
collect2: error: ld returned 1 exit status
make[3]: *** [f951] Error 1
../.././riscv-gcc/gcc/fortran/Make-lang.in:97: recipe for target 'f951' failed
make[3]: Leaving directory '/home/jason/work/riscv-gnu-toolchain/build-gcc-linux-stage2/gcc'
make[2]: *** [all-gcc] Error 2
Makefile:4316: recipe for target 'all-gcc' failed
make[2]: Leaving directory '/home/jason/work/riscv-gnu-toolchain/build-gcc-linux-stage2'
make[1]: *** [all] Error 2
Makefile:951: recipe for target 'all' failed
make[1]: Leaving directory '/home/jason/work/riscv-gnu-toolchain/build-gcc-linux-stage2'
make: *** [stamps/build-gcc-linux-stage2] Error 2
Makefile:308: recipe for target 'stamps/build-gcc-linux-stage2' failed

jim-wilson, 2020-05-08:
This looks like a versioning problem.  It appears that the mpfr version you have is either too old or too new to work with the gcc version you have.  By default, we only build C and C++ for the elf compiler, but we build C, C++, and Fortran for the linux compiler, and it is the Fortran build that is failing, so only the linux compiler is failing to build.

I build on Ubuntu 16.04 often, but the gcc I have is gcc-5.4.  Maybe your system hasn't been updated in a long time.  If so, updating the OS may fix it.  You can find the exact OS version in /etc/os-release.  I have 16.04.06.

The other main alternative would be to add compatible gmp/mpc/mpfr version sources into the gcc tree.  If you cd into riscv-gcc, and run "./contrib/download_prerequisties" you will get compatible versions of them downloaded which should solve the build problem.

Or you could work around the problem by disabling the Fortran support if you don't need it.  In the toplevel Makefile.in look for
    --enable-languages=c,c++,fortran
and change it to
    --enable-languages=c,c++
You have to do this twice.

jasonlee1001, 2020-05-08:
Thank you very much for replying.
My OS version is 16.04.06,too.
I installed gcc 4.8.3 because of EDA compatible issue.
I did installed gmp/mpc/mpfr manually in order to install gcc-4.8.3 successfully.
 
In fact, I'm currently to do the same work of this -> https://github.com/cnrv/fpga-rocket-chip
In section 1.3 Preparing the Project, they said "install elf-gcc and linux-gcc and set RISCV variable in advance" , so I think it may need linux compiler.

 "./contrib/download_prerequisties" , I will do this to check the required versions. 
Thanks again.

jim-wilson, 2020-05-08:
Another option then would be to try doing CC=gcc-5 when configuring and making, though I don't know for sure if this will work.  In theory it should force everything to use gcc-5 which is more likely to work than gcc-4.8.
CC=gcc-5 configure ...
CC=gcc--5 make ...

In general, the bigger the difference between the gcc versions, the more likely you are to run into problems, as new gcc versions might be using C/C++ features that old gcc versions don't support, or might trigger bugs in old gcc versions that weren't found before we stopped supporting it.

jim-wilson, 2020-05-08:
I keep forgetting about C++.  So this would have to be
CC=gcc-5 CXX=g++-5 configure ...
CC=gcc-5 CXX=g++-5 make ...

adel15, 2020-05-08:
Hi All,
I have error on building rv32imc.
I used Ubuntu 18.04 and gcc/g++ are 7.5.0.

/riscv-gnu-toolchain$ **./configure --prefix=/opt/riscv  --with-arch=rv32imc**
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /bin/grep
checking for fgrep... /bin/grep -F
checking for grep that handles long lines and -e... (cached) /bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
/riscv-gnu-toolchain$ **sudo make linux**
....
make[3]: Leaving directory '/home/adel/SweRV/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc'
make[2]: Leaving directory '/home/adel/SweRV/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc'
make[1]: Leaving directory '/home/adel/SweRV/riscv-gnu-toolchain/build-gcc-linux-stage1'
make -C build-gcc-linux-stage1 inhibit-libc=true install-target-libgcc
make[1]: Entering directory '/home/adel/SweRV/riscv-gnu-toolchain/build-gcc-linux-stage1'
/bin/bash .././riscv-gcc/mkinstalldirs /opt/riscv /opt/riscv
make[2]: Entering directory '/home/adel/SweRV/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc'
/bin/bash ../../.././riscv-gcc/libgcc/../mkinstalldirs /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0
/usr/bin/install -c -m 644 libgcc.a /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/
chmod 644 /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/libgcc.a
/opt/riscv/riscv32-unknown-linux-gnu/bin/ranlib /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/libgcc.a
/usr/bin/install -c -m 644 libgcov.a /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/
chmod 644 /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/libgcov.a
/opt/riscv/riscv32-unknown-linux-gnu/bin/ranlib /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/libgcov.a
parts="crtbegin.o crtbeginS.o crtbeginT.o crtend.o crtendS.o crtbegin.o crtend.o crti.o crtn.o crtendS.o crtbeginT.o";				\
for file in $parts; do					\
  rm -f /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/$file;		\
  /usr/bin/install -c -m 644 $file /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/;	\
  case $file in 					\
    *.a)						\
      /opt/riscv/riscv32-unknown-linux-gnu/bin/ranlib ../.././gcc/$file ;;	\
  esac;							\
done
/bin/bash ../../.././riscv-gcc/libgcc/../mkinstalldirs /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/include
/usr/bin/install -c -m 644 unwind.h /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/include
/bin/bash ../../.././riscv-gcc/libgcc/../mkinstalldirs /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/include
/usr/bin/install -c -m 644 ../../.././riscv-gcc/libgcc/gcov.h /opt/riscv/lib/gcc/riscv32-unknown-linux-gnu/9.2.0/include
make[2]: Leaving directory '/home/adel/SweRV/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu/libgcc'
make[1]: Leaving directory '/home/adel/SweRV/riscv-gnu-toolchain/build-gcc-linux-stage1'
mkdir -p stamps/ && touch stamps/build-gcc-linux-stage1
rm -rf stamps/build-glibc-linux-rv32imc-ilp32 build-glibc-linux-rv32imc-ilp32
mkdir build-glibc-linux-rv32imc-ilp32
cd build-glibc-linux-rv32imc-ilp32 && \
	CC="riscv32-unknown-linux-gnu-gcc " \
	CXX="riscv32-unknown-linux-gnu-g++ " \
	CFLAGS="  -mcmodel=medlow -g -O2 " \
	CXXFLAGS="  -mcmodel=medlow -g -O2 " \
	ASFLAGS=" -mcmodel=medlow " \
	/home/adel/SweRV/riscv-gnu-toolchain/riscv-glibc/configure \
	--host=riscv32-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/home/adel/SweRV/riscv-gnu-toolchain/linux-headers/include \
	--disable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv32-unknown-linux-gnu-gcc 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether riscv32-unknown-linux-gnu-gcc  accepts -g... yes
checking for gcc... gcc
checking for riscv32-unknown-linux-gnu-readelf... riscv32-unknown-linux-gnu-readelf
checking whether we are using the GNU C++ compiler... no
checking whether riscv32-unknown-linux-gnu-g++  accepts -g... no
checking whether riscv32-unknown-linux-gnu-g++  can link programs... no
checking for sysdeps preconfigure fragments... aarch64 alpha arm csky hppa i386 m68k microblaze mips nios2 powerpc riscv glibc requires the A extension
Makefile:238: recipe for target 'stamps/build-glibc-linux-rv32imc-ilp32' failed
make: *** [stamps/build-glibc-linux-rv32imc-ilp32] Error 1

jim-wilson, 2020-05-08:
Like the message says, riscv glibc requires the A extension.  Without the A extension, you can't have any synchronization across processes, which means you can't have a multi-core multi-process multi-programmed multi-threaded OS like linux.  There are probably some single process embedded real time OS ports that can work without the A extension, but not linux.  On a single core single process system you can emulate atomics by disabling interrupts.

adel15, 2020-05-08:
Thank you for your help.
I fixed it with: 
./configure --prefix=/opt/riscv  --with-arch=rv32imac

jasonlee1001, 2020-05-09:
I reinstalled Ubuntu 16.04 on Virtual Box VM.
Its default gcc version is 5.4, and it can make Linux compiler without errors.
So gcc-4.8.3 seems not work to make Linux compiler.
I will use Jim’s recommendation to use “CC=gcc-5 CXX=g++-5” before ./configure and make again.
Thank you Jim!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-05-10 by jasonlee1001