riscv-collab/riscv-gnu-toolchain #633: linker error

andrewavram34, 2020-05-11:
Hello!
I have installed MSYS2 on windows 10 and followed this steps to compile the riscv-gnu-toolchain on my computer -> https://www.emdalo.com/posts/risc-v-gnu-compiler-toolchain-howto-compile-on-windows/

All went well until I tried to make the helloworld program. 
The linker does not seem to find crt0.o and othe errors appear.

---------------------------
$ riscv32-unknown-elf-gcc.exe hello.c -o hello.elf
c:/msys64/opt/riscv/bin/../lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld.exe: cannot find crt0.o: No such file or directory
c:/msys64/opt/riscv/bin/../lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld.exe: cannot find crtbegin.o: No such file or directory
c:/msys64/opt/riscv/bin/../lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld.exe: cannot find -lgcc
c:/msys64/opt/riscv/bin/../lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld.exe: cannot find -lc
c:/msys64/opt/riscv/bin/../lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld.exe: cannot find -lgloss
collect2.exe: error: ld returned 1 exit status
--------------------------

I searched for crt0.o  on all the folders but didn't seem to find it. 
Can someone please help me ?
Thank you very much, in advance!

jim-wilson, 2020-05-11:
The crt0.o file should be in $prefix/riscv32-unknown-elf/lib.  If it isn't there, then maybe the build did not run to completion.  There should have been an error reported in that case.  Look in the make.err file for an error.

pfaelectrique, 2021-11-10:
> The crt0.o file should be in $prefix/riscv32-unknown-elf/lib. If it isn't there, then maybe the build did not run to completion. There should have been an error reported in that case. Look in the make.err file for an error.

i have the same problem : i used the musl gcc wrapper after i installed the musl  libraray ( i choose the target : riscv64-unknown-elf) but when i did the compilation i had the following error :
/usr/local/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: cannot find crtbeginS.o: No such file or directory
/usr/local/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: cannot find -lgloss
collect2: error: ld returned 1 exit status
Can u help me ??

jim-wilson, 2021-11-10:
We only have musl support for the linux target.  You can't build a riscv64-uknown-elf toolchain with musl.  Missing libraries suggests that something went wrong during the build.  Try looking at the build log.  Or just try the build again.  If you still have the build dirs then it will restart the part that failed.

pfaelectrique, 2021-11-11:
thank you @jim-wilson 
i still have problems , my goal is  to use the cross compiler   riscv-64-unknown-elf  and make a static link  with the musl library ( using a thread application ).i have installed musl with the target   riscv-64-unknown-elf  and then i linked staticlly musl but i have problems with crt0.o file of the compiler .  i did these commands :
riscv64-unknown-elf-gcc  -c -I/home/mariem/musl_novembre/include  thread.c
riscv64-unknown-elf-gcc  -L/home/mariem/musl_novembre/lib   -static -lc  thread.o
and i had the following error :
/usr/local/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: /usr/local/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib/crt0.o: in function `.L0 ':
(.text+0x2a): undefined reference to `__libc_init_array'
collect2: error: ld returned 1 exit status
I think this error is beacause the crt0.o file is compatible with libc.a file of glib c not with the libc.a file of musl
Mr @jim-wilson   do have any idea about this ?? i need  your advice

pfaelectrique, 2021-11-11:
thank you @jim-wilson
i still have problems , my goal is to use the cross compiler riscv-64-unknown-elf and make a static link with the musl library ( using a thread application ).i have installed musl with the target riscv-64-unknown-elf and then i linked staticlly musl but i have problems with crt0.o file of the compiler . i did these commands :
riscv64-unknown-elf-gcc -c -I/home/musl/include thread.c
riscv64-unknown-elf-gcc -L/home/musl/lib -static -lc thread.o

and i had the following error :
/usr/local/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: /usr/local/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib/crt0.o: in function .L0 ': (.text+0x2a): undefined reference to __libc_init_array'
collect2: error: ld returned 1 exit status

I think this error is beacause the crt0.o file is compatible with libc.a file of glib c not with the libc.a file of musl
Mr @jim-wilson do have any idea about this ?? i need your advice

jim-wilson, 2021-11-11:
When compiling gcc, it does some tests to determine the capabilitiles of the target C library, and then adjusts its code to be compatible.  So you can't build a toolchain that uses newlib, substitute musl instead, and then expect that to work.  It is better to properly configure a toolchain that uses musl before you build it.  We don't have build rules for that.  But you can try copying the riscv64-uknown-elf build rules and changing them to use musl instead of newlib, like the musl linux rules do.  I don't know if musl supports embedded targets like newlib does though.

libc_init_array is a function defined in newlib.  The newlib crt0.o knows that the newlib C library defines this function, and so will call it.  So yes, if you are changing the C library, you need to change the crt0.o also.

TommyMurphyTM1234, 2022-09-27:
This is an issue with the user's approach to building the toolchain and attempting to switch standard libraries. Not an issue with the RISC-V GCC toolchain itself. Closing it as it's not a valid issue.

closed 2022-09-27 by TommyMurphyTM1234