riscv-collab/riscv-gnu-toolchain #635: Getting configure: error: C compiler cannot create executables error

thakureie, 2020-05-14:
We have two GCC versions 4 & 6 however we could see error while installing this. Please help. ``

Getting configure: error: C compiler cannot create executables error 

 cat /etc/redhat-release
CentOS release 6.10 (Final)

uname -a
Linux XXX 2.6.32-754.28.1.el6.x86_64 #1 SMP Wed Mar 11 18:38:45 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

GCC Version :6

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/x86_64-pc-linux-gnu/6.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: ../configure
Thread model: posix
gcc version 6.1.0 (GCC)

GCC 4

/usr/bin/gcc -v
Using built-in specs.
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-languages=c,c++,objc,obj-c++,java,fortran,ada --enable-java-awt=gtk --disable-dssi --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-1.5.0.0/jre --enable-libgcj-multifile --enable-java-maintainer-mode --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --disable-libjava-multilib --with-ppl --with-cloog --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.4.7 20120313 (Red Hat 4.4.7-23) (GCC)

export RISCV=/usr/bin/riscv
/root/riscv-isa-sim-master/configure --prefix=$RISCV --enable-commitlog --enable-misaligned
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking for gcc... gcc
checking whether the C compiler works... no
configure: error: in `/usr/local':
configure: error: C compiler cannot create executables
See `config.log' for more details

`[root@asic-5 local]# cat config.log
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by RISC-V ISA Simulator configure ?, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /root/riscv-isa-sim-master/configure --prefix=/usr/bin/riscv --enable-commitlog --enable-misaligned

## --------- ##
## Platform. ##
## --------- ##

hostname = asic-5.enphaseenergy.com
uname -m = x86_64
uname -r = 2.6.32-754.28.1.el6.x86_64
uname -s = Linux
uname -v = #1 SMP Wed Mar 11 18:38:45 UTC 2020

/usr/bin/uname -p = unknown
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /usr/lib64/qt-3.3/bin
PATH: /usr/share/lsf/9.1/linux2.6-glibc2.3-x86_64/etc
PATH: /usr/share/lsf/9.1/linux2.6-glibc2.3-x86_64/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /sbin
PATH: /bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /opt/dell/srvadmin/bin
PATH: /opt/dell/srvadmin/sbin
PATH: /root/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2127: checking build system type
configure:2141: result: x86_64-unknown-linux-gnu
configure:2161: checking host system type
configure:2174: result: x86_64-unknown-linux-gnu
configure:2393: checking for gcc
configure:2409: found /usr/local/bin/gcc
configure:2420: result: gcc
configure:2649: checking for C compiler version
configure:2658: gcc --version >&5
gcc (GCC) 6.1.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2669: $? = 0
configure:2658: gcc -v >&5
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/x86_64-pc-linux-gnu/6.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: ../configure
Thread model: posix
gcc version 6.1.0 (GCC)
configure:2669: $? = 0
configure:2658: gcc -V >&5
gcc: error: unrecognized command line option '-V'
gcc: fatal error: no input files
compilation terminated.
configure:2669: $? = 1
configure:2658: gcc -qversion >&5
gcc: error: unrecognized command line option '-qversion'; did you mean '--version'?
gcc: fatal error: no input files
compilation terminated.
configure:2669: $? = 1
configure:2689: checking whether the C compiler works
configure:2711: gcc    conftest.c  >&5
/usr/bin/ld: crtbegin.o: No such file: No such file or directory
collect2: error: ld returned 1 exit status
configure:2715: $? = 1
configure:2753: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "RISC-V ISA Simulator"
| #define PACKAGE_TARNAME "spike"
| #define PACKAGE_VERSION "?"
| #define PACKAGE_STRING "RISC-V ISA Simulator ?"
| #define PACKAGE_BUGREPORT "Andrew Waterman"
| #define PACKAGE_URL ""
| /* end confdefs.h.  */
|
| int
| main ()
| {
|
|   ;
|   return 0;
| }
configure:2758: error: in `/usr/local':
configure:2760: error: C compiler cannot create executables
See `config.log' for more details

## ---------------- ##
## Cache variables. ##
## ---------------- ##

ac_cv_build=x86_64-unknown-linux-gnu
ac_cv_env_CCC_set=
ac_cv_env_CCC_value=
ac_cv_env_CC_set=
ac_cv_env_CC_value=
ac_cv_env_CFLAGS_set=
ac_cv_env_CFLAGS_value=
ac_cv_env_CPPFLAGS_set=
ac_cv_env_CPPFLAGS_value=
ac_cv_env_CXXCPP_set=
ac_cv_env_CXXCPP_value=
ac_cv_env_CXXFLAGS_set=
ac_cv_env_CXXFLAGS_value=
ac_cv_env_CXX_set=
ac_cv_env_CXX_value=
ac_cv_env_LDFLAGS_set=
ac_cv_env_LDFLAGS_value=
ac_cv_env_LIBS_set=
ac_cv_env_LIBS_value=
ac_cv_env_STOW_PREFIX_set=
ac_cv_env_STOW_PREFIX_value=
ac_cv_env_STOW_ROOT_set=
ac_cv_env_STOW_ROOT_value=
ac_cv_env_build_alias_set=
ac_cv_env_build_alias_value=
ac_cv_env_host_alias_set=
ac_cv_env_host_alias_value=
ac_cv_env_target_alias_set=
ac_cv_env_target_alias_value=
ac_cv_host=x86_64-unknown-linux-gnu
ac_cv_prog_ac_ct_CC=gcc

## ----------------- ##
## Output variables. ##
## ----------------- ##

AR=''
CC='gcc'
CFLAGS=''
CPPFLAGS=''
CXX=''
CXXCPP=''
CXXFLAGS=''
DEFS=''
DTC=''
ECHO_C=''
ECHO_N='-n'
ECHO_T=''
EGREP=''
EXEEXT=''
GREP=''
HAVE_INT128=''
INSTALL_DATA=''
INSTALL_PROGRAM=''
INSTALL_SCRIPT=''
LDFLAGS=''
LIBOBJS=''
LIBS=''
LTLIBOBJS=''
OBJEXT=''
PACKAGE_BUGREPORT='Andrew Waterman'
PACKAGE_NAME='RISC-V ISA Simulator'
PACKAGE_STRING='RISC-V ISA Simulator ?'
PACKAGE_TARNAME='spike'
PACKAGE_URL=''
PACKAGE_VERSION='?'
PATH_SEPARATOR=':'
RANLIB=''
SHELL='/bin/sh'
STOW_PREFIX=''
STOW_ROOT=''
ac_ct_CC='gcc'
ac_ct_CXX=''
bindir='${exec_prefix}/bin'
build='x86_64-unknown-linux-gnu'
build_alias=''
build_cpu='x86_64'
build_os='linux-gnu'
build_vendor='unknown'
datadir='${datarootdir}'
datarootdir='${prefix}/share'
docdir='${datarootdir}/doc/${PACKAGE_TARNAME}'
dvidir='${docdir}'
enable_stow=''
exec_prefix='NONE'
host='x86_64-unknown-linux-gnu'
host_alias=''
host_cpu='x86_64'
host_os='linux-gnu'
host_vendor='unknown'
htmldir='${docdir}'
includedir='${prefix}/include'
infodir='${datarootdir}/info'
libdir='${exec_prefix}/lib'
libexecdir='${exec_prefix}/libexec'
localedir='${datarootdir}/locale'
localstatedir='${prefix}/var'
mandir='${datarootdir}/man'
oldincludedir='/usr/include'
pdfdir='${docdir}'
prefix='/usr/bin/riscv'
program_transform_name='s,x,x,'
psdir='${docdir}'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
subprojects=''
subprojects_enabled=''
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "RISC-V ISA Simulator"
#define PACKAGE_TARNAME "spike"
#define PACKAGE_VERSION "?"
#define PACKAGE_STRING "RISC-V ISA Simulator ?"
#define PACKAGE_BUGREPORT "Andrew Waterman"
#define PACKAGE_URL ""

configure: exit 77
[root@asic-5 local]#
`

jim-wilson, 2020-05-14:
The config.log says
configure:2393: checking for gcc
configure:2409: found /usr/local/bin/gcc
and that is apparently your gcc-6, so it appears that your gcc-6 install is broken.  Have you tried compiling a hello world program with gcc-6?  Does that work?  I suspect it won't.  Where did this local gcc-6 install come from?  You will have to fix that, or remove it, or remove /usr/local/bin from your PATH.

I see that you are on Centos 6.  The best way to get a newer gcc version on Centos 6 is to use the Software Collections.  See
https://www.softwarecollections.org/en/docs/
And then look at the available toolset packages, e.g. devtoolset-6
https://www.softwarecollections.org/en/scls/rhscl/devtoolset-6/
They also have gcc-7 and gcc-8 packages available.

TommyMurphyTM1234, 2022-09-27:
This is a build environment issue and not an issue with the RISC-V GCC toolchain itself. Closing as invalid.

closed 2022-09-27 by TommyMurphyTM1234