riscv-collab/riscv-gnu-toolchain #636: Does riscv-gdb close python by default

Phantom1003, 2020-05-17:
I run riscv64-unknown-linux-gnu-gdb with gef, got the following info:
```
GNU gdb (GDB) 8.3.0.20190516-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
/home/phantom/.gdbinit:1: Error in sourced command file:
/home/phantom/.gdbinit-gef.py:58: Error in sourced command file:
Undefined command: "from".  Try "help".
```
And I try to run some python code in gdb, I got
```
>Scripting in the "Python" language is not supported in this copy of GDB.
```
Which option should open to support gdb ? Thanks

jim-wilson, 2020-05-17:
You probably don't have the necessary OS packages installed to support python.  If you look at the build-gdb-linux/gdb/config.log file you will see checks for python support, and probably one of them failed.  Note that you may need to have a python development package installed in order for gdb to find python header files and libraries.  That is the part most likely missing.  You probably have the pythoin package installed but not the python developer package.  This varies from one distro to the next.

Phantom1003, 2020-05-17:
I find info in `build-gdb-linux/gdb/config.log`:
```
configure:10550: checking for python2.7
configure:10568: gcc -o conftest -g -O2  -I/usr/include/python2.7 -I/usr/include/python2.7 -static-libstdc++ -static-libgcc  conftest.c -lpython2.7 -lpthread -ldl -lutil -lm -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions -lm -ldl  >&5
conftest.c:53:10: fatal error: Python.h: No such file or directory
 #include "Python.h"
          ^~~~~~~~~~
compilation terminated.
```
And I run the command
```
sudo apt-get install python-dev  
```
Although I am still compiling, I think this should work, thanks

Phantom1003, 2020-05-17:
By the way, you should also use a python2 version gef.

closed 2020-05-17 by Phantom1003