riscv-collab/riscv-gnu-toolchain #637: -mno-relax seems not working for gas

quantrpeter, 2020-05-17:
temp.s:
```
.section .text
.globl _start
_start:
	    beq x0,x0,0x1234
```

riscv64-unknown-elf-as -march=rv32i -mno-relax temp.s -o temp.o 
riscv64-unknown-elf-objdump -ds temp.o

```
temp.o:     file format elf32-littleriscv

Contents of section .text:
 0000 63140000 6f000000                    c...o...        
Contents of section .riscv.attributes:
 0000 41190000 00726973 63760001 0f000000  A....riscv......
 0010 05727633 32693270 3000               .rv32i2p0.      

Disassembly of section .text:

00000000 <_start>:
   0:	00001463          	bnez	zero,8 <_start+0x8>
   4:	0000006f          	j	4 <_start+0x4>
```

**_beq x0,x0,0x1234_** still become "**_bnez	zero,8 <_start+0x8>_**" and "**_j	4 <_start+0x4>_**"

Thanks

jim-wilson, 2020-05-17:
We must still emit correct code, even when -mno-relax is used.  -mno-relax just disables link time optimizations.

beq 0x1234 says to branch to an absolute address of 0x1234.  Since we don't know what address the linker will assign to this code, we have no choice but to use a long jump for it.  If it isn't in range of a jump at link time, then the linker will give an error.

If you don't want a long jump, then branch to a label within range.  Alternatively, you could use .+0x234 to branch to an offset from the current instruction, but note that branches have a 12-bit offset, so .+0x1234 is not in range, and will still branch around a jump.

Nelson1225, 2020-05-18:
> We must still emit correct code, even when -mno-relax is used. -mno-relax just disables link time optimizations.

Thanks for @jim-wilson  pointing it out, this is the key point :)  For now there are five RISC-V link time relaxations,

1. LUI + LO pattern used to get/load/store a 32-bit address into register.
lui (R_RISCV_HI20)
addi/lw/sw (R_RISCV_LO12_I/S)

may be relaxed to ->

addi/lw/sw with gp as based register (R_RISCV_GPREL_I/S)

2. Get/load/store a 32/64 bits address by pc-relative patterns.
auipc (R_RISCV_PCREL_HI20)
addi/lw/sw (R_RISCV_PCREL_LO12_I/S)

may be relaxed to ->

addi/lw/sw with gp as based register (R_RISCV_GPREL_I/S)

3. CALL/TAIL function
auipc (R_RISCV_CALL)
jalr

may be relaxed to ->

jal (R_RISCV_JAL)

4. TLS LE pattern
5. Alignment issue

I'm not sure if the alignment issue belongs to the one of relaxations, but I usually think so for RISC-V.   So just like Jim mentioned, `-mno-relax` just disables link time relaxations mentioned above, except alignment issue, since we must still emit correct code.  Once you disable the link time relaxations, assembler won't generate `R_RISCV_RELAX` relocation to the instruction which could be relaxed.  Linker only do the relaxation for the instructions which have attached `R_RISCV_RELAX`.

quantrpeter, 2020-05-18:
> We must still emit correct code, even when -mno-relax is used. -mno-relax just disables link time optimizations.
>
> beq 0x1234 says to branch to an absolute address of 0x1234. Since we don't know what [...]

you are correct, using label solve this problem. But can we define the offset of a label? I tried ".equ label1,0x12" but not working.

jim-wilson, 2020-05-18:
Note that branch instructions always take an address as an argument, and then convert that address to an offset to write into the instruction encoding.  There are no instructions that accept a label offset, except as a special case as part of relocation syntax.  This works the same for all architectures, arm, x86, etc, not just RISC-V.  So label offsets are not very useful.  label offsets are even more of a problem on RISC-V, because with the aggressive link time relaxation, text label offsets aren't constants, and hence not usable in most contexts.  Making them work with linker relaxation turned off would require adding a lot of special case code to the assembler which is unlikely to be worth the effort considering how rarely they would be useful.

But if you do really want a label offset, you can get it via label subtraction.  The special symbol "." (dot) is the current address.  So you can do something like this

```
rohan:2024$ cat tmp.s
	.data
	.word (label1-.)
	.skip 10
label1:
rohan:2025$ riscv32-unknown-elf-as -o tmp.o tmp.s
rohan:2026$ riscv32-unknown-elf-objdump --section=.data --full-contents ./a.out

./a.out:     file format elf32-littleriscv

Contents of section .data:
 0000 0e000000 00000000 00000000 0000      ..............  
rohan:2027$ 
```

The data section has the value 0xe (14) which is the size of .word (4) plus the skipped bytes (10).

quantrpeter, 2020-06-22:
thanks all

closed 2020-06-22 by quantrpeter