riscv-collab/riscv-gnu-toolchain #638: can I avoid qemu installation?

neelgala, 2020-05-18:
I am looking at reducing build time.

The submodule clone for qemu takes significantly long. 
Is it possible to avoid qemu  clone/installation completely. 

Is qemu required for building rest of the tools?

jim-wilson, 2020-05-18:
qemu is optional.  it is used for testing the toolchain (make check) which is important to the toolchain maintainers.  For end users, it is also useful for running programs, but not needed if you want to use another simulator or want to use hardware.  qemu is only built if you use make check.  But it gets downloaded by default if you do a submodule update and don't specify what modules to update.  I don't know if there is an easy solution for that.  Maybe just comment out the qemu entry in the .gitmodules file before you do the submodule update?  This assumes you do a regular clone first, edit the .gitmodules file, and then do a git submodule update --init --recursive.

neelgala, 2020-05-19:
Thanks. Wanted a confirmation if I had to change scripts for that.
Closing Issue.

closed 2020-05-19 by neelgala