riscv-collab/riscv-gnu-toolchain #641: undefined reference to `__fixdfsi'

shuaitq, 2020-05-24:
Hi,

I have generated a toolchain with
```
./configure --with-abi=ilp32 --with-arch=rv32i --prefix=/home/tom/git/riscv-gnu-toolchain/build
```

I try to compile this C++ code

```
const int finish_value = 0x00020000;
const int char_value = 0x00010000;

volatile int * const tohost                 = (int * const)0x40008000;
volatile int * const led                    = (int * const)0x40008004;
volatile int * const switch_address         = (int * const)0x40008008;
volatile int * const button                 = (int * const)0x4000800C;
volatile int * const seven_segment_display  = (int * const)0x40008010;
volatile int * const vga_port_start         = (int * const)0x40009000;
volatile int * const vga_port_end           = (int * const)0x40044FFC;

// NOTE: compile with g++ filename.cpp -std=c++11

#include <cmath>
#define _sq(x) ((x)*(x)) // square
#define _cb(x) abs((x)*(x)*(x)) // absolute value of cube
#define _cr(x) (unsigned char)(pow((x),1.0/3.0)) // cube root

unsigned char GR(int,int);
unsigned char BL(int,int);

unsigned char RD(int i,int j){
return (char)(_sq(cos(atan2(j-512,i-512)/2))*255);
}

unsigned char GR(int i,int j){
return (char)(_sq(cos(atan2(j-512,i-512)/2-2*acos(-1)/3))*255);
}

unsigned char BL(int i,int j){
return (char)(_sq(cos(atan2(j-512,i-512)/2+2*acos(-1)/3))*255);
}
volatile int * addr = vga_port_start;
void pixel_write(int,int);
extern "C" void start(){
for(int j=0;j<240;j++)
for(int i=0;i<256;i++)
{
pixel_write(i,j);
++addr;
}
for(;;);
}
void pixel_write(int i, int j){
static unsigned char color[3];
color[0] = RD(i,j)&255;
color[1] = GR(i,j)&255;
color[2] = BL(i,j)&255;
int val = ((color[0] & 0xf0) << 4) | (color[1] & 0xf0) | ((color[2] & 0xf0) >> 4);
*addr = val;
}
```
using 
```
	riscv32-unknown-elf-g++ -march=rv32i -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -lgcc -L /home/tom/git/riscv-gnu-toolchain/build/lib/gcc/riscv32-unknown-elf/9.2.0/ -lm -L /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib/ -O3 -c c1.cpp -o t.o
	riscv32-unknown-elf-ld -Ttest.lds -lgcc -L /home/tom/git/riscv-gnu-toolchain/build/lib/gcc/riscv32-unknown-elf/9.2.0/ -lm -L /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib/ -O3 -o test.elf
```
test.lds
```
SECTIONS 
{
    . = 0x00000000;

    .text :
    {
        start.o
        t.o
    }
}
 ```

start.o is compiled from
```
    lui	sp,0x1
    j   start
```
and I got a lot of errors
```
riscv32-unknown-elf-g++ -march=rv32i -mabi=ilp32 -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles -lgcc -L /home/tom/git/riscv-gnu-toolchain/build/lib/gcc/riscv32-unknown-elf/9.2.0/ -lm -L /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib/ -O3 -c c1.cpp -o t.o
riscv32-unknown-elf-ld -Ttest.lds -lgcc -L /home/tom/git/riscv-gnu-toolchain/build/lib/gcc/riscv32-unknown-elf/9.2.0/ -lm -L /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib/ -O3 -o test.elf
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-e_atan2.o): in function `.L12':
e_atan2.c:(.text+0xc0): undefined reference to `__divdf3'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-e_rem_pio2.o): in function `.L4':
e_rem_pio2.c:(.text+0x130): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: e_rem_pio2.c:(.text+0x170): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: e_rem_pio2.c:(.text+0x1bc): undefined reference to `__eqdf2'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-e_rem_pio2.o): in function `.L6':
e_rem_pio2.c:(.text+0x2cc): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-k_rem_pio2.o): in function `__kernel_rem_pio2':
k_rem_pio2.c:(.text+0x84): undefined reference to `__divsi3'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x274): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x2a8): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x338): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x3b8): undefined reference to `__eqdf2'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x6a8): undefined reference to `__eqdf2'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x6e8): undefined reference to `__gedf2'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x708): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x744): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0x784): undefined reference to `__gedf2'
riscv32-unknown-elf-ld: k_rem_pio2.c:(.text+0xcec): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-s_atan.o): in function `.L7':
s_atan.c:(.text+0x28c): undefined reference to `__gtdf2'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-s_atan.o): in function `.L6':
s_atan.c:(.text+0x370): undefined reference to `__divdf3'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-s_atan.o): in function `.L19':
s_atan.c:(.text+0x394): undefined reference to `__divdf3'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-k_cos.o): in function `__kernel_cos':
k_cos.c:(.text+0x58): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-k_sin.o): in function `__kernel_sin':
k_sin.c:(.text+0x58): undefined reference to `__fixdfsi'
riscv32-unknown-elf-ld: start.o:(.text+0x4): undefined reference to `start'
riscv32-unknown-elf-ld: /home/tom/git/riscv-gnu-toolchain/build/riscv32-unknown-elf/lib//libm.a(lib_a-s_floor.o): in function `floor':
s_floor.c:(.text+0x80): undefined reference to `__gtdf2'
riscv32-unknown-elf-ld: s_floor.c:(.text+0x13c): undefined reference to `__gtdf2'
riscv32-unknown-elf-ld: s_floor.c:(.text+0x16c): undefined reference to `__gtdf2'
```
I am very confused and do not know how to solve it

jim-wilson, 2020-05-24:
The linker requires multiple obscure options in the correct order to function.  It is generally unwise to try to call the linker yourself.  The compiler driver gcc knows how to call the linker properly and using gcc to link gives the best result.  Otherwise, if you really want to call ld to link, then use gcc -v to see how the gcc driver is calling the linker, and then edit the linker command as necessary.

In this case, the problem is that functions are only pulled from a library if they are needed.  You are linking with libgcc first, and then with libm, but libm needs some functions from libgcc.  So when libgcc was processed by the linker, the divdf3 function was not needed, so it wasn't pulled in.  Then you link with libm which calls divdf3 and now it is needed, but it wasn't pulled in earlier, so you get an error.  You can fix this by moving the -lgcc after the -lm.  But of course if libgcc needs functions from libm or libc, this won't work.  This can be fixed by linking with libgcc twice, e.g. -lgcc -lm -lgcc.  Worst case, there are multiple circular dependencies and you need to mention each library multiple times.  The linker has a short hand for this which is --start-group -lm -lgcc --end-group and now the two libraries will be searched repeatedly until no new functions are pulled from either file.

Your linker script looks odd.  Including the start.o and t.o object files in the .text section will force all sections in those files to be put in the text section in the output, which is very unlikely to be correct, and may not work.  It is best to put file names on the linker command line. But if you want to include them in the linker script, use e.g. INPUT(start.o t.o) and put that before the SECTION command.

TommyMurphyTM1234, 2022-09-27:
Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234