riscv-collab/riscv-gnu-toolchain #642: Forcing a different starting base address while compiling RISCV instructions

vineetkotariya, 2020-05-26:
I am trying to cross-compile my riscv code and trying to set the base address of the .text section of my program to a certain address (BASE_ADDR). However, I am facing some issues.

I used the command:
`$ riscv64-unknown-elf-gcc -march=rv64gcv -Ttext BASE_ADDR -o hello_world.riscv hello_world.s
`
I want to set BASE_ADDR = 80000000

But the compiler throws up the following error: 

> /path/lib/gcc/riscv64-unknown-elf/7.2.0/crtbegin.o: In function `deregister_tm_clones':
crtstuff.c:(.text+0x0): relocation truncated to fit: R_RISCV_HI20 against `__TMC_LIST__'
/path/lib/gcc/riscv64-unknown-elf/7.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: Symbol needs debug section which does not exist
collect2: error: ld returned 1 exit status

(The command works fine as long as my addresses don't cross 80000000. For example, the compiler works fine when BASE_ADDR=70000000)

I used the following code:

```
.file  "hello_world.s"
.option nopic
.text
.align 1
.globl main
.type  main, @function
main:
    csrr x4, mhartid
    slli x4, x4, 3
    add x3, x3, x4
    nop
    nop
```

Am I doing something wrong here or missing something?
Thanks

aswaterman, 2020-05-26:
For RV64, the address 8000_0000 is out of range of the LUI instruction,
which can only load addresses between approx. -2^31 and +2^31.

You could use -mcmodel=medany (which will use AUIPC instead of LUI, getting
around this limitation), but your C library also needs to be compiled with
that flag, or else the LUIs in your C library will remain problematic.

jim-wilson, 2020-05-26:
Palmer Dabbelt wrote a blog post a while ago that talks about code models.
https://www.sifive.com/blog/all-aboard-part-4-risc-v-code-models

You can use --with-cmodel to choose the default code model when configuring the toolchain.  SiFive compilers are built with --with-cmodel=medany.

vineetkotariya, 2020-05-27:
Thanks! It worked for me.

I used this command for configuring my toolchain (and then built it) to make it the default code model:

`$ ../configure --with-cmodel=medany --prefix=$RISCV
`
And then used this command for compiling:

`$ riscv64-unknown-elf-gcc -march=rv64gcv -Ttext 80000000 -mcmodel=medany -o hello_world.riscv hello_world.s
`

closed 2020-05-27 by vineetkotariya