riscv-collab/riscv-gnu-toolchain #643: Cannot find libsanitizer

wmmc88, 2020-05-28:
When compiling a simple main using `riscv64-unknown-elf-gcc`, I encounter a linker error where the compiler cannot find ubsan. Is ubsan not compatible with baremetal riscv64 targets?

Main.cc
```
int main(void){
	volatile int i = 1;
	int k = 0x7fffffff;
	k += i;
	return 0;	
}
```

Command Line Invocation
```
riscv64-unknown-elf-gcc -fsanitize=undefined main.cc
```

Error:
```
riscv64-unknown-elf/bin/ld: cannot find -lubsan
collect2: error: ld returned 1 exit status
```

jim-wilson, 2020-05-28:
The sanitizers use syscalls like mmap, so they can only be built for targets running an OS, like linux or freebsd.

wmmc88, 2020-05-28:
Hmm... Would it still be possible to just provide my own definitions of the ubsan handlers that avoid syscalls?

jim-wilson, 2020-05-28:
I don't know offhand if anyone has ever tried this.  If you look at riscv-gcc/libsanitizer/configure.tgt you will see a list of supported systems, and everything else sets UNSUPPORTED=1.  So add two lines to match riscv*-*-elf* and do nothing, or maybe disable the default case that sets UNSUPPORTED at the end, and try building a toolchain.  You will likely run into a lot of problems.

fwsGonzo, 2020-05-30:
I have written an ubsan implementation here: https://github.com/fwsGonzo/barebones/blob/master/src/crt/ubsan.c

It's not magic, but you aren't going to get much out of it unless you have the ability to turn an address into a symbol+offset. And yes, all the ubsans are mostly alike, and there are different versions. You can choose to support whatever you like. Each time you write code that can fail in a certain way the compiler will generate an ubsan-failure-function for it, and you will know what to do.

You should pass `-fsanitize=undefined -fno-sanitize=vptr` as implementing the vptr table is just make-work, unless you really need it.

Good luck

TommyMurphyTM1234, 2022-09-29:
No update/activity in over two years.
@jim-wilson has explained the reason why the issue arises with the bare metal toolchain.
@fwsGonzo has provided a possible alternative for this case.

closed 2022-09-29 by TommyMurphyTM1234