riscv-collab/riscv-gnu-toolchain #644: Cross-Compiled Dynamically-Linked executable not found in qemu

chaozju, 2020-06-06:
Hi I am working on cross compilation for [wasmer](https://github.com/wasmerio/wasmer) to RISC-V.

Things work fine until I want to execute it. 
May I get some guidance on this ? Sry I am still a green hand on cross compiling.
```bash
[root@stage4 ~]# ls
ffi  wasmer
[root@stage4 ~]# chmod 777 ./wasmer
[root@stage4 ~]# file wasmer
wasmer: ELF 64-bit LSB shared object, UCB RISC-V, version 1 (SYSV), dynamically linked, interpreter /lib/ld.so.1, with debug_info, not stripped
[root@stage4 ~]# ldd wasmer
        linux-vdso.so.1 (0x000000200001a000)
        libffi.so.8 => /lib64/lp64d/libffi.so.8 (0x000000200007b000)
        libstdc++.so.6 => /lib64/lp64d/libstdc++.so.6 (0x0000002000084000)
        libutil.so.1 => /lib64/lp64d/libutil.so.1 (0x00000020001f3000)
        libdl.so.2 => /lib64/lp64d/libdl.so.2 (0x00000020001f7000)
        librt.so.1 => /lib64/lp64d/librt.so.1 (0x00000020001fb000)
        libpthread.so.0 => /lib64/lp64d/libpthread.so.0 (0x0000002000202000)
        libgcc_s.so.1 => /lib64/lp64d/libgcc_s.so.1 (0x000000200021d000)
        libc.so.6 => /lib64/lp64d/libc.so.6 (0x0000002000231000)
        /lib/ld.so.1 => /lib/ld-linux-riscv64-lp64d.so.1 (0x0000002000000000)
        libm.so.6 => /lib64/lp64d/libm.so.6 (0x000000200033e000)
[root@stage4 ~]# ./wasmer
-bash: ./wasmer: No such file or directory
```

jim-wilson, 2020-06-06:
Linux compilers are hard.  riscv-gnu-toolchain is fine for building linux binaries to run on a simulator, but should not be used to build a cross compiler for a linux target.  This is unlikely to work.  It might work if you static link the binaries.

If you have a system running linux, then the best way to build a cross compiler for it is to make a sysroot, which contains a copy of the target /usr/include, /usr/lib, /lib and whatever else might be necessary for the compiler such as lib64 dirs if the target has them.  Then use the sysroot when building binutils and gcc by passing the --with-sysroot=  option pointing at a copy of target sysroot.

The "No such file or dir" error would usually mean that you are missing the dynamic linker, but ldd says it is there, so I don't know what is wrong here offhand.

The "root@stage4" looks wrong.  It appears you are using an old and obsolete fedora linux distro.  If you are following
https://wiki.qemu.org/Documentation/Platforms/RISCV
then stop, because this info is hopelessly out-of-date.  The up-to-date info is here
https://fedoraproject.org/wiki/Architectures/RISC-V/Installing

TommyMurphyTM1234, 2022-09-27:
No RISC-V GCC toolchain issue identified.

closed 2022-09-27 by TommyMurphyTM1234