riscv-collab/riscv-gnu-toolchain #648: riscv64-oe-linux-gcc Dynamic loader is failing

ghalib210, 2020-06-09:
Hello,

          I have build an yocto project Standard SDK with openembedded-core & meta-riscv sources, build was success. With this SDK build i got cross-compiled Riscv Toolchain, with this Tool i have compiled an helloworld source code compile successfully but when i a have copied this build to another machine, dynamic loader failing and getting an Error like,
$ riscv64-oe-linux-gcc --version
$ riscv64-oe-linux-gcc: No such file or directory 

$ ldd riscv64-oe-linux-gcc
.	linux-vdso.so.1 (0x00007ffd379c0000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f305c209000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f305be18000)
	**/home/khan/openembedded-core/riscv-build/tmp-glibc/deploy/sdk/sdk/sysroots/x86_64-oesdk-linux/lib/ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2 (0x00007f305c5a7000)**

if you see here the path is hardcoded, Is any option or commands to give custom PATH.

Thanks,
Ghalib khan.

jim-wilson, 2020-06-09:
Linux compilers are hard.  riscv-gnu-toolchain isn't designed to produce compilers to run on the target.   It would be best to avoid riscv-gnu-toolchain if you want a compiler that runs on the target, and build a toolchain the traditional way. 

However, all of this is pointless, as openembedded has support for building a toolchain for the target OS, so you should use that instead of going outside openembdded.  Just enable the target toolchain package and openembedded will build it.  I'm not an openembedded expert so I don't know how to do that offhand, but the sifive meta-sifive has a demo-coreip-cli build target that builds an OS with a native toolchain.  So using that might be an easier option.

TommyMurphyTM1234, 2022-09-27:
Question answered.

closed 2022-09-27 by TommyMurphyTM1234