riscv-collab/riscv-gnu-toolchain #651: Issue with Zvediv extension isn't enabled

soniab, 2020-06-14:
I have installed latest riscv-gnu-toolchain which has 
                  GNU assembler (GNU Binutils) 2.34
                  riscv64-unknown-elf-gcc (GCC) 9.2.0 

I am doing assembly programming for riscv-v extension instruction set architecture. 

While compiling assembly program, I am getting following issue:
"Warning: vediv is set but Zvediv extension isn't enabled"

I am using "../configure --prefix=$HOME/local/riscvv09/gnu --enable-multilib" to configure riscv-gnu-toolchain. 

It will be helpful if you can help me with what options I need to use to enable Zvediv extension.

kito-cheng, 2020-06-14:
Use -march to enable that, append Zvediv to the end of arch string, e.g: `riscv64-unknown-elf-gcc -march=rv64gcv_zvediv`

soniab, 2020-06-14:
Thank you for your reply. 

Does riscv-gnu toolchain supports half precision floating point (float16) datatype?   Do I need to provide any parameter while configuring riscv-gnu toolchain?

Nelson1225, 2020-06-15:
> Does riscv-gnu toolchain supports half precision floating point (float16) datatype?

No, fp16 is not yet supported in riscv-gnu-toolchain.  But I think it will be eventually.

soniab, 2020-06-16:
Thank you for your reply.

closed 2020-06-19 by kito-cheng