riscv-collab/riscv-gnu-toolchain #652: qemu git submodule init failed for unadvertised object

ailinzhou, 2020-06-16:
when init git submodule, **qemu** module failed for unadvertised object.
`$ git submodule update --init --recursive`
```
error: Server does not allow request for unadvertised object 57dfc2c4d51e770ed3f617e5d1456d1e2bacf3f0
Fetched in submodule path 'qemu', but it did not contain 
57dfc2c4d51e770ed3f617e5d1456d1e2bacf3f0. Direct fetching of that commit failed.```

jim-wilson, 2020-06-16:
Maybe a temporary problem.  The commit does exist in the qemu tree and I was able to check it out just now.

You can incidentally build the toolchain without qemu.  qemu is used for optional testing, i.e. make check, and is only built if you run make check or make stamps/build-qemu.  And it is useful for end users that want to be able to run the code they compile if they don't have hardware.

ailinzhou, 2020-06-16:
The problem is the parent model `riscv-gnu-toolchain`  now is at  qemu submodel  ``57dfc2c`` commit id. But this commit(57dfc2c) in `qemu` upstream repository was deleted. 
@jim-wilson maybe we can update the `qemu` let it point to the new commit id or just `git pull` in qemu model to point to origin/master. 
When someone got the some error, workaround way is comment out the qemu sub module. Follow the steps below.
* go in directory of riscv-gnu-toolchain
* open .gitmodules file
* add # at begin of qemu model
   ```
   #    [submodule "qemu"]
   #	     path = qemu
   #	     url = https://git.qemu.org/git/qemu.git
   ```
* run `git submodule update --init --recursive` at directory of riscv-gnu-toolchain.
* then follow the riscv-gnu-toolchain's README continue you job

jim-wilson, 2020-06-16:
I can still see the commit there.
https://git.qemu.org/?p=qemu.git;a=commit;h=57dfc2c4d51e770ed3f617e5d1456d1e2bacf3f0

Unfortunately an update to qemu 5 breaks the rv32-linux support.  The rv32 glibc patches are not upstream yet.  There is a patch series submitted upstream to correct this which has y2038 changes which make it incompatible with the version currently in riscv-gnu-toolchain.  So we need both glibc and qemu updated together, plus fixing the fallout from the updates, which is non trivial work, and may not happen for a while.

patrickod, 2020-08-12:
> Maybe a temporary problem. The commit does exist in the qemu tree and I was able to check it out just now.

Bumping this thread as I encountered the same issues this evening getting my own environment working.

```
[patrickod@finn:~/code/riscv-gnu-toolchain]$ git submodule update --init
error: Server does not allow request for unadvertised object 57dfc2c4d51e770ed3f617e5d1456d1e2bacf3f0
Fetched in submodule path 'qemu', but it did not contain 57dfc2c4d51e770ed3f617e5d1456d1e2bacf3f0. Direct fetching of that commit failed.
```

@jim-wilson are you still able to retrieve this commit? If so I would be very interested to compare our configurations/commands used as it's not clear to what the issue is. The git web link you shared is still working also :thinking:

jim-wilson, 2020-08-12:
Yes, I can still check it out.  And the commit id is still obviously there.  There is a link to it above.  If a clone/update fails, I would suggest waiting and trying it again.  git servers do sometimes fail.

jedevc, 2020-10-08:
For people having this issue, any chance you were doing a shallow clone with `--depth=<low number>`?

Apparently some git servers "don't like that", so won't let you, see [here](https://superuser.com/questions/1240216/what-does-the-git-error-message-server-does-not-allow-request-for-unadvertised). Doing a full clone seemed to resolve the problem for me, guess you do learn something new every day :smile:

diadatp, 2020-12-19:
@jedevc I have encountered the same problem and can confirm your observation about shallow clones. Initially I tried to perform a shallow clone of the submodules, which was the qemu git server didn't like.

```
error: Server does not allow request for unadvertised object 57dfc2c4d51e770ed3f617e5d1456d1e2bacf3f0
Fetched in submodule path 'qemu', but it did not contain 57dfc2c4d51e770ed3f617e5d1456d1e2bacf3f0. Direct fetching of that commit failed.
```

This is the workaround I used:

```
git clone https://github.com/riscv/riscv-gnu-toolchain.git
cd riscv-gnu-toolchain/
git submodule update --progress --init qemu/
git submodule update --progress --depth=1 --init
```

This brings the total download size down to around 0.5 GB.

Misc info:
```
git --version 
git version 2.17.1
```

Perhaps the shallow key can be specified in the .gitmodules file for the non qemu repositories?

jim-wilson, 2020-12-19:
Kito recently made a change to add spike (aka riscv-isa-sim) which doesn't git clone it until you need it.  I suggested we do the same thing for qemu.  We just haven't had a chance to try that yet.  If we do that, and it works, then we don't need to git clone qemu unless you specify it with a make check.  See the spike-src and pk-src makefile rules for instance.

TommyMurphyTM1234, 2022-09-29:
No update/activity in almost two years and the issue may be sure to the "shallow clone" mentioned above. And, in the intervening time, the build script has changed and the issue hasn't been reported by others.

closed 2022-09-29 by TommyMurphyTM1234