riscv-collab/riscv-gnu-toolchain #656: How can I reconfigure and remake the toolchain?

kingstone1927, 2020-06-17:
I tried to reconfigure and remake the toolchain, but it did not work. What should I do if I want to reconfigure and remake the toolchain? Also, how can I uninstall the old toolchain? 

Thanks,

WilliamWangPeng, 2020-06-17:
you should first,  then configure, prefix if you want to install in whereever you want, choose the march and mabi if you need.
``` 
$make clean  

$./configure --prefix=/home/u/tools/rv64 --with-arch=rv64gc --with-abi=lp64  
```

WilliamWangPeng, 2020-06-17:
> I tried to reconfigure and remake the toolchain, but it did not work. What should I do if I want to reconfigure and remake the toolchain? Also, how can I uninstall the old toolchain?
> 
> Thanks,

or you can upload the shootscreen and let us see more details about your problems when remaking and reconfiguring

jim-wilson, 2020-06-18:
There is no easy way to uninstall the old toolchain.  I suggest using a different prefix for every install, so that you can remove an old one by using rm -rf on the install tree.

kingstone1927, 2020-06-20:
> There is no easy way to uninstall the old toolchain. I suggest using a different prefix for every install, so that you can remove an old one by using rm -rf on the install tree.

I was able to use `make clean` to rebuild the toolchain. I think it will be helpful it the toolchain has like `make uninstall` functionality, so we can just uninstall it with one command.

Thank you so much for helping me.

kingstone1927, 2020-06-20:
> you should first, then configure, prefix if you want to install in whereever you want, choose the march and mabi if you need.
> 
> ```
> $make clean  
> 
> $./configure --prefix=/home/u/tools/rv64 --with-arch=rv64gc --with-abi=lp64  
> ```

I did the exact thing you said, and I was able to rebuild the toolchain. I am still stuck at how I can efficiently remove it from the system. Thanks so much for your help.

jim-wilson, 2020-06-20:
riscv-gnu-toolchain contains multiple independent separately maintained packages.  So it isn't possible for riscv-gnu-toolchain to have an uninstall command unless each package has its own uninstall command.  And they don't.

There is also the problem that if you install version X, and then try to use the version Y uninstall command, it is very unlikely to work, as they don't install/uninstall the same files.  So if you want to uninstall version X, you would need the original version X sources, which you may not have.

The only easy way to make this work is to use some packaging scheme like rpm which tracks the files that get installed, and hangs around so that when you want to uninstall it you still have the file list.  This is beyond the scope of the riscv-gnu-toolchain project.  This is just a simple build script for convenience and nothing more.

kingstone1927, 2020-06-25:
> riscv-gnu-toolchain contains multiple independent separately maintained packages. So it isn't possible for riscv-gnu-toolchain to have an uninstall command unless each package has its own uninstall [...]

Thanks you for the detailed answer.

TommyMurphyTM1234, 2022-09-27:
There are several different options for building the toolchain these days so I am closing this issue.

closed 2022-09-27 by TommyMurphyTM1234