riscv-collab/riscv-gnu-toolchain #657: Does riscv-gnu-toolchain supports vector extension?

WilliamWangPeng, 2020-06-17:
Hi everyone
    I wonder if riscv-gnu-toolchain supports the vector extension, I use the following command but the errors occur, saying "Fatal error: -march=rv32gcv: unsupported ISA subset `v'"  .  
    
```  
./configure --prefix=/home/u/tools/rv32 --with-arch=rv32gcv --with-abi=ilp32    
  ```    

[image]    

Best Regards
William

kito-cheng, 2020-06-18:
Vector extension support not hold in master branch, there are several branch prefixed with `rvv` in riscv-gnu-toolchain, you can checkout corresponding branch to build vector extension enabled toolchain.

e.g.
rvv-0.7.x
rvv-0.7.1
rvv-0.8.x
rvv-0.9.x

jim-wilson, 2020-06-18:
The existing publicly available toolchains only contain binutils support though, and a small gcc patch to accept v in the arch string.

WilliamWangPeng, 2020-06-19:
thank you so much, I installed and get the vector support now, but riscv32-unknown-elf-gcc doesn't work well with  
1 spike (which also supports vector but doesn't know how to enable vector support, spike --isa=RV32IV or --isa=RV32IFADMC  command line doesn't work)    
2 pk (which can support vector when building, ../configure --prefix=$RISCV -host=riscv32-unknown-elf ,  but the pk actually is in $RISCV/riscv64-unknown-elf, not in $RISCV/riscv32-unknown-elf, I do install riscv32-unknown-elf and riscv32-unknown-elf-gcc works well )   
do you have some ideas where I need to modify?

WilliamWangPeng, 2020-06-19:
> The existing publicly available toolchains only contain binutils support though, and a small gcc patch to accept v in the arch string.

Hi dear Jim
    I have installed riscv32-unknown-elf   
$wget https://mirror.iscas.ac.cn/plct/riscv-gnu-toolchain.tar.bz2
$tar -jxvf riscv-gnu-toolchain.tar.bz2
$ cd riscv-gnu-toolchain
$ git checkout rvv-0.8.x
$ ./configure --prefix=/home/u/tools/test --with-arch=rv32gcv --with-abi=ilp32
$ make
$ export PATH=home/u/tools/test/bin:$PATH  
but when I use riscv32-unknown-elf-gcc to compile the c program to elf, spike pk elf has no results.  here are my modified supports for spike and pk. 
1 spike (which also supports vector but doesn't know how to enable vector support, spike --isa=RV32IV or --isa=RV32IFADMC command line doesn't work)
2 pk (which can support vector when building, ../configure --prefix=$RISCV -host=riscv32-unknown-elf , but the pk actually is in $RISCV/riscv64-unknown-elf, not in $RISCV/riscv32-unknown-elf, I do install riscv32-unknown-elf and riscv32-unknown-elf-gcc works well )  
in PK
machine/encoding.h 
@@ -14,6 +14,7 @@
#define MSTATUS_MPP         0x00001800
+#define MSTATUS_VS          0x01800000
#define MSTATUS_FS          0x00006000 

machine/minit.c 
@@ -27,6 +27,10 @@ static void mstatus_init()
  if (supports_extension('F'))
    write_csr(mstatus, MSTATUS_FS);

  +// Enable vector extension
  +if (supports_extension('V'))
  +  write_csr(mstatus, MSTATUS_VS);

jim-wilson, 2020-06-19:
I've never looked at spike vector support.  spike is not part of riscv-gnu-toolchain.

I've never looked at pk vector support.  pk is not part of riscv-gnu-toolchain.  You might need to configure pk specailly to build a riscv32 version of it.  Try looking at the pk sources to see how it supports riscv64 versus riscv32.

WilliamWangPeng, 2020-06-22:
> I've never looked at spike vector support. spike is not part of riscv-gnu-toolchain.
>
> I've never looked at pk vector support. pk is not part of riscv-gnu-toolchain. You might need to configure pk [...]

Thank you dear Jim
    I have solved the problem in riscv32-unknown-elf-gcc in linking our intrinsic vsetvl, I added the defination of instruction delcaration,  
+/* RVV */
+DECLARE_INSN(vsetvl, MATCH_VSETVL, MASK_VSETVL)
+DECLARE_INSN(vsetvli, MATCH_VSETVLI, MASK_VSETVLI)

riscv-gnu-toolchain/riscv-binutils/include/opcode/riscv-opc.h
/* RVV */
#define MATCH_VSETVL  0x80007057
#define MASK_VSETVL   0xfe00707f
#define MATCH_VSETVLI 0x00007057
#define MASK_VSETVLI  0x8000707f

Also there is one problem at rvv-0.9.x, the riscv-v-spec 0.9 has already changed the fcsr to vcsr , but rvv-0.9.x has not changed and remained still.
 
[image]

WilliamWangPeng, 2020-06-22:
after changing the gnu toolchain rvv-0.9.x, I use riscv32-unknown-elf-gcc to link the test.o to elf, and use qemu-riscv32 elf, which runs well, and use riscv32-unknown-elf-objdump -d elf, we can see the intrinsic vsetvl vadd and vsub.

jim-wilson, 2020-06-22:
fcsr is for the F (and D) extension.  vcsr is for the V extension.  The toolchain supports both registers.

WilliamWangPeng, 2020-06-23:
> fcsr is for the F (and D) extension. vcsr is for the V extension. The toolchain supports both registers.

thank you Jim, you are always so kind and helpful

closed 2020-08-03 by WilliamWangPeng