riscv-collab/riscv-gnu-toolchain #658: Suggestion about change the fcsr to vcsr in rvv-0.9.x

WilliamWangPeng, 2020-06-22:
Hi dear everyone 
    Recently I discover rvv-0.9.x branch in riscv-gnu-toolchain didn't support the real riscv-v-spec 0.9, cause you can see at riscv-binutils/include/opcode/riscv-opc.h, fcsr which has changed to vscr in riscv-v-spec 0.9 is still defined there, at the same time,  vlenb doesn't define in this branch. I hope riscv can add the definition of these registers which can follow the riscv-v-spec 0.9. 

/* RVV */
DECLARE_CSR(vstart, CSR_VSTART)
DECLARE_CSR(vxsat, CSR_VXSAT)
DECLARE_CSR(vxrm, CSR_VXRM)
DECLARE_CSR(vl, CSR_VL)
DECLARE_CSR(vtype, CSR_VTYPE)

#ifdef DECLARE_CSR
DECLARE_CSR(frm, CSR_FRM)
DECLARE_CSR(fcsr, CSR_FCSR)

DECLARE_CSR(fcsr, CSR_FCSR)

#define CSR_FCSR 0x3

[image]

kito-cheng, 2020-06-22:
Thanks your report, @Nelson1225 could you take a look for this?

Nelson1225, 2020-06-22:
> Thanks your report, @Nelson1225 could you take a look for this?

OK I can take care of this.

Nelson1225, 2020-06-22:
Hi @WilliamWangPeng 

> Recently I discover rvv-0.9.x branch in riscv-gnu-toolchain didn't support the real riscv-v-spec 0.9, 

Umm that's weird...  I remember there is only one difference - The rvv constraints for mask.

`The destination vector register group for a masked vector instruction cannot overlap the source mask register (v0), unless the destination vector register is being written with a mask value (e.g., comparisons) or the scalar result of a reduction. Otherwise, an illegal instruction exception is raised.`

That means the mask overlap constraints should be more strict in rvv-0.9.x.  But this won't cause problems if we check the constraints loosely in the assembler.

> cause you can see at riscv-binutils/include/opcode/riscv-opc.h, fcsr which has changed to vscr in riscv-v-spec 0.9 is still defined there, at the same time, vlenb doesn't define in this branch. I hope riscv can add the definition of these registers which can follow the riscv-v-spec 0.9.

The vcsr isn't the alias name for fcsr.  In fact, they are not the same CSR.  I notice that the code you gave seems like the old implementation, maybe rvv-0.8.x, rather than rvv-0.9.x.  Unless I bump riscv-binutils to the wrong commit...  I can check this later.  See the following link, 
https://github.com/riscv/riscv-binutils-gdb/blob/rvv-0.9.x/include/opcode/riscv-opc.h#L2928

Besides, the binutils 2.35 will be released recently (7/4).  The current rvv-0.9.x is based on the binutils-2.34 release.  There are some changes for priv specs and CSR on upstream, so probably need to update the rvv, too.  I plan to update the rvv-1.0.x to the binutils-2.35, since rvv-0.9.x and rvv-1.0.x should be fully compatible.

Thanks
Nelson

jim-wilson, 2020-06-22:
The fcsr register was not dropped from the F (and D) extension so it still exists.  Both the new vcsr and vlenb registers are there.  Maybe you forgot to update submodules after switching branches?  Try "git submodule update --init --recursive".

Stone66666, 2020-06-23:
退订

Stone66666, 2020-06-23:
能不给我发邮件了不，我要退订

kito-cheng, 2020-06-23:
@Stone66666 You can click the` unwatch` button and select `not watching` to unsubscribe, 你可以點擊上面的 unwatch 按鈕後選擇 `not watching` 退訂

closed 2020-08-03 by WilliamWangPeng