riscv-collab/riscv-gnu-toolchain #659: How to clone required version of tools gcc, binutiles & gilibc

ghalib210, 2020-06-23:
Hello, 

          can any one know, how to clone required version's of tools gcc, binutiles & gilibc.

Thanks,
khan.

jim-wilson, 2020-06-23:
Depends on which versions you want.  If you want the default versions provided by riscv-gnu-toolchain, then you just clone riscv-gnu-toolchain (the green clone or download button will give you the URL you need), and then you can use "git submodule update --init --recursive" to get the tool submodules.  This is explained in the README.md file by the way.

ghalib210, 2020-06-23:
I know how to clone latest version, see for example i want an 8.3 version of gcc, then how can i clone it??

jim-wilson, 2020-06-23:
We haven't been tracking individual gcc versions.  But you can run "git log > tmp.file" and then look for the entry that mentions bumping gcc to gcc 9, and then checkout the last commit before that.  We use merge commits, so there will actually be two commits for the gcc 9 bump.  Just checking real quick I see that the last commit before the gcc bump to gcc 9 is a binutils bump which goes with the gcc 9 bump, so you might want to go two more commits backward.

closed 2022-09-27 by TommyMurphyTM1234