riscv-collab/riscv-gnu-toolchain #660: how can I modify functionality of an instruction?

kingstone1927, 2020-06-25:
[image]

I have been trying to find ways to add custom instructions to the RISC-V cross-compiler and received  great answers from the contributors of the RISC-V-gnu-toolchain. I am looking at this chart that I found online. I wonder if there is any way I can modify the functionality/behavior of one of these instructions? or how by looking at these instruction can I know their functionality/behaviors (ADD, SUB, AND...)? what tool in the toolchain takes these instructions and perform the logical behaviors of it?

Thanks in advance.

kito-cheng, 2020-06-29:
Not sure what's your goal, but maybe you want to adding new instruction? if so I would suggest you take a look for [riscv-opcdes](https://github.com/riscv/riscv-opcodes) and [riscv-isa-sim](https://github.com/riscv/riscv-isa-sim), it would be a good start to using riscv-opcdes to add your instruction to spike.

And then you might want using your new instruction, it will require toolchain changes which is complicated, so I would suggest your using `.insn` directive[1] .

[1] https://sourceware.org/binutils/docs-2.34/as/RISC_002dV_002dDirectives.html#index-INSN-directives

pcotret, 2020-07-15:
Hi,
my goal is pretty much the same: play with my new instruction in Spike and then make my compiler aware of it.
Do you confirm that the "standard" way to add new instructions in the RISC-V compiler is to use directives?

kito-cheng, 2020-07-16:
> Do you confirm that the "standard" way to add new instructions in the RISC-V compiler is to use directives?

It's not "standard" way to add new instructions to compiler, but it's easier way to evaluate your instruction without modify assembler & compiler.

The better way to me is add instruction to assembler & compiler, however it's little complicated to do, because both LLVM and GCC are pretty complicate, but I won't say that's standard way :P

kito-cheng, 2020-10-12:
Close this issue, seems issue resolved and long time no further discussion/response, feel free to create issue or reopen this issue if you have further question.

closed 2020-10-12 by kito-cheng

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2

NazerkeT, 2023-01-18:
Hi @kito-cheng and all!

Thanks for making support of the .insn directive! Though, I have some confusion: which of these instruction group rules([riscv ](https://sourceware.org/binutils/docs-2.34/as/RISC_002dV_002dFormats.html#RISC_002dV_002dFormats)vs [s390](https://sourceware.org/binutils/docs-2.30/as/s390-Formats.html#s390-Formats)) should we follow? I think you mentioned both of these rv64 and s390 groups, however they look a bit different, for example rv64 does not have rx, while s390 does have one. Could you please clarify it? I think i got a bit confused :(

Thanks,
Best wishes,

manasabatchu, 2024-03-25:
I am adding all packed simd instructions to riscv-opc.c file what are instructions have the immediate value of 5 bits are working fine and what are instructions have 3bits and 4bits values I am unable to add to the riscv-opc.c file inplace of operands what type of operands can I use to solve this problem ( I am using Toolchain version is 0.9.2) can any one have idea about it please help me

TommyMurphyTM1234, 2024-03-25:
> I am adding all packed simd instructions to riscv-opc.c file what are instructions have the immediate value of 5 bits are working fine and what are instructions have 3bits and 4bits values I am [...]

You should probably review the content in this and other linked issues (see above) and maybe ask about how to customise the toolchain for custom instructions in the upstream GCC/Binutils dorums/mailing lists.

> ( I am using Toolchain version is 0.9.2)

What do you mean by this? The GCC toolchain compiled from this repo these days is GCC 13!

* https://github.com/riscv-collab/riscv-gnu-toolchain/blob/9a28c809a5d355874def06a414aa5f272fae564d/.gitmodules#L5-L8