riscv-collab/riscv-gnu-toolchain #661: Linker Error Unrecognised Emulation Mode: abi=ilp32

VizorKit, 2020-06-25:
This is my gcc version.
```
Using built-in specs.
COLLECT_GCC=C:\msys64\opt\riscv\bin\riscv64-unknown-elf-gcc.exe
COLLECT_LTO_WRAPPER=c:/msys64/opt/riscv/bin/../libexec/gcc/riscv64-unknown-elf/10.1.0/lto-wrapper.exe
Target: riscv64-unknown-elf
Configured with: /home/chris/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 10.1.0 (GCC)
```
Here is multi lib support output
```
.;
rv32i/ilp32;@march=rv32i@mabi=ilp32
rv32im/ilp32;@march=rv32im@mabi=ilp32
rv32iac/ilp32;@march=rv32iac@mabi=ilp32
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imafc/ilp32f;@march=rv32imafc@mabi=ilp32f
rv64imac/lp64;@march=rv64imac@mabi=lp64
rv64imafdc/lp64d;@march=rv64imafdc@mabi=lp64d
```
Here is the complete script I am running.
```
riscv64-unknown-elf-as -march=rv32imac -mabi=ilp32 novectors.s -o novectors.o
riscv64-unknown-elf-gcc -march=rv32imac -mabi=ilp32 -Wall -O2 -nostdlib -nostartfiles -ffreestanding  -c notmain.c -o notmain.o
riscv64-unknown-elf-ld -march=rv32imac -mabi=ilp32 novectors.o notmain.o -T memmap -o notmain.elf
riscv64-unknown-elf-objdump -D notmain.elf > notmain.list
riscv64-unknown-elf-objcopy notmain.elf -O ihex notmain.hex
riscv64-unknown-elf-objcopy notmain.elf -O binary notmain.bin 
```
and here is the error output.
```
novectors.s: Assembler messages:
novectors.s: Warning: end of file not at end of a line; newline inserted
C:\msys64\opt\riscv\bin\riscv64-unknown-elf-ld.exe: unrecognised emulation mode: abi=ilp32
Supported emulations: elf64lriscv elf32lriscv
C:\msys64\opt\riscv\bin\riscv64-unknown-elf-objdump.exe: 'notmain.elf': No such file
C:\msys64\opt\riscv\bin\riscv64-unknown-elf-objcopy.exe: 'notmain.elf': No such file
C:\msys64\opt\riscv\bin\riscv64-unknown-elf-objcopy.exe: 'notmain.elf': No such file
```
I figure I will include the entire contents of the file as well. Incase the linker error is because of the sections/code I have laid out.

> novectors.S
```
.globl _start
_start:
    lui sp, 0x80004 # Initialize the stack
    jal notmain     # jump to main
    j   .           # unsure what . does.

.globl _dummy
_dummy:
    ret
.globl _put_32
_put_32:
    sw a1, (a0)
    ret
.globl _get_32
_get_32:
    lw a0, (a0)
    ret
```
> notmain.c
```
void _put_32(unsigned int, unsigned int);
unsigned int _get_32(unsigned int);
void _dummy(unsigned int);

#define GPIOBASE            0x10012000
#define GPIO_VALUE          (GPIOBASE+0x00)
#define GPIO_INPUT_EN       (GPIOBASE+0x04)
#define GPIO_OUTPUT_EN      (GPIOBASE+0x08)
#define GPIO_PORT           (GPIOBASE+0x0C)
#define GPIO_PUE            (GPIOBASE+0x10)
#define GPIO_OUT_XOR        (GPIOBASE+0x40)

int notmain ( void )
{
    unsigned int rx;

    _put_32(GPIO_OUTPUT_EN,(1<<19)|(1<<21)|(1<<22));
    _put_32(GPIO_PORT,(1<<19)|(1<<21)|(1<<22));
    _put_32(GPIO_OUT_XOR,0);
    while(1)
    {
        _put_32(GPIO_PORT,(1<<19)|(1<<21)|(1<<22));
        for(rx=0;rx<2000000;rx++) _dummy(rx);
        _put_32(GPIO_PORT,0);
        for(rx=0;rx<2000000;rx++) _dummy(rx);
    }

    return(0);
}
```
> memmap
```
MEMORY
{
    rom : ORIGIN = 0x20010000, LENGTH = 0x1000
    ram : ORIGIN = 0x80000000, LENGTH = 0x4000
}
SECTIONS
{
    .text : { *(.text*) } > rom
    .rodata : { *(.rodata*) } > rom
    .bss : { *(.bss*) } > ram
}
```

VizorKit, 2020-06-25:
I was able to resolve this by doing quite a bit of research into how the flags for emulation work with linker. the correct ld.exe options are.

```
riscv64-unknown-elf-ld -melf32lriscv novectors.o notmain.o -T memmap -o notmain.elf
```

jim-wilson, 2020-06-25:
There are a number of obscure linker options that need to be used for correct output.  Knowledge of these options is encoded into gcc.  So the easiest way to link is to use gcc instead of calling ld directly.

VizorKit, 2020-06-25:
sorry last question. When I get the output of `notmain.list` at the bottom this section is produced.

```
Disassembly of section .riscv.attributes:

00000000 <.riscv.attributes>:
   0:	2a41                	jal	190 <_start-0x2000fe70>
   2:	0000                	unimp
   4:	7200                	flw	fs0,32(a2)
   6:	7369                	lui	t1,0xffffa
   8:	01007663          	bgeu	zero,a6,14 <_start-0x2000ffec>
   c:	0020                	addi	s0,sp,8
   e:	0000                	unimp
  10:	1004                	addi	s1,sp,32
  12:	7205                	lui	tp,0xfffe1
  14:	3376                	fld	ft6,376(sp)
  16:	6932                	flw	fs2,12(sp)
  18:	7032                	flw	ft0,44(sp)
  1a:	5f30                	lw	a2,120(a4)
  1c:	326d                	jal	fffff9c6 <notmain+0xdffef9b2>
  1e:	3070                	fld	fa2,224(s0)
  20:	615f 7032 5f30      	0x5f307032615f
  26:	30703263          	0x30703263
	...

Disassembly of section .comment:

00000000 <.comment>:
   0:	3a434347          	fmsub.d	ft6,ft6,ft4,ft7,rmm
   4:	2820                	fld	fs0,80(s0)
   6:	29554e47          	fmsub.s	ft8,fa0,fs5,ft5,rmm
   a:	3120                	fld	fs0,96(a0)
   c:	2e30                	fld	fa2,88(a2)
   e:	2e31                	jal	32a <_start-0x2000fcd6>
  10:	0030                	addi	a2,sp,8
```
Is this actually being included in my hex or bin output? It seems wild. Those instructions were no where in my code, and all flags i provided tell me that there should be no other libraries linked.

What is this section? They don't correspond to real addresses either, so I'm guessing its just the compilers comments?

jim-wilson, 2020-06-26:
These sections don't contain code.  The problem here is that you used objdump -D.  You should use objdump -d instead.  And complain to whatever idiot told you to use objdump -D that this is wrong.  objdump -D is only useful in obscure situations, and should only be used by experts.

No, these sections aren't included in the hex or bin output as they are not loadable sections.  They just contain useful info.  You can tell that they aren't loadable as they have VMA and LMA of zero in the objdump -h output.  Or alternatively, they don't have the A flag in the readelf -A output.

The comment section contains strings.  You can view it with "objdump --full-contets --section=.comment a.out".

The riscv_attributes section contains flag bits to indicate what features are used by this executable.  You can view it with "riscv64-unknown-elf-readelf -A a.out".  These can be used by the linker to detect when incompatible object files are linked together.

TommyMurphyTM1234, 2022-09-27:
Questions relating to tools usage as opposed to issues with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234