riscv-collab/riscv-gnu-toolchain #663: Build gfortran Native(RISCV) toolchain.

ghalib210, 2020-06-29:
Hello,

I want to build an gfortran Native(RISCV) Toolchain on U540, For that i have followed this step's,

1. cloned riscv-gnu sources 
2. cd riscv-gnu-toolchain/riscv-gcc
3. mkdir build install
4. cd build && ../configure --prefix=$PWD/../install --enable-languages=c,c++,gfortran
5. make

Here Build got failed, Here is the log

/home/root/khan/riscv-gcc/build/./gcc/xgcc -B/home/root/khan/riscv-gcc/build/./gcc/ -B/home/root/khan/riscv-gcc/build/../install/riscv64-unknown-linux-gnu/bin/ -B/home/root/khan/riscv-gcc/build/../install/riscv64-unknown-linux-gnu/lib/ -isystem /home/root/khan/riscv-gcc/build/../install/riscv64-unknown-linux-gnu/include -isystem /home/root/khan/riscv-gcc/build/../install/riscv64-unknown-linux-gnu/sys-include    -g -O2 -march=rv32imac -mabi=ilp32 -O2  -g -O2 -DIN_GCC    -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -fPIC -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector   -fPIC -I. -I. -I../../../.././gcc -I../../../../../libgcc -I../../../../../libgcc/. -I../../../../../libgcc/../gcc -I../../../../../libgcc/../include  -DHAVE_CC_TLS  -o _negdi2.o -MT _negdi2.o -MD -MP -MF _negdi2.dep -DL_negdi2 -c ../../../../../libgcc/libgcc2.c -fvisibility=hidden -DHIDE_EXPORTS
In file included from /usr/include/sys/cdefs.h:460,
                 from /usr/include/features.h:450,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:27,
                 from ../../../../../libgcc/../gcc/tsystem.h:87,
                 from ../../../../../libgcc/libgcc2.c:27:
/usr/include/bits/wordsize.h:28:3: error: #error "rv32i-based targets are not supported"
 # error "rv32i-based targets are not supported"
   ^~~~~
In file included from /usr/include/bits/long-double.h:19,
                 from /usr/include/sys/cdefs.h:461,
                 from /usr/include/features.h:450,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:27,
                 from ../../../../../libgcc/../gcc/tsystem.h:87,
                 from ../../../../../libgcc/libgcc2.c:27:
/usr/include/bits/wordsize.h:28:3: error: #error "rv32i-based targets are not supported"
 # error "rv32i-based targets are not supported"
   ^~~~~
In file included from /usr/include/sys/cdefs.h:461,
                 from /usr/include/features.h:450,
                 from /usr/include/bits/libc-header-start.h:33,
                 from /usr/include/stdio.h:27,
                 from ../../../../../libgcc/../gcc/tsystem.h:87,
                 from ../../../../../libgcc/libgcc2.c:27:
/usr/include/bits/long-double.h:40:10: fatal error: bits/long-double-32.h: No such file or directory
 #include <bits/long-double-32.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [Makefile:494: _negdi2.o] Error 1
make[4]: Leaving directory '/home/root/khan/riscv-gcc/build/riscv64-unknown-linux-gnu/lib32/ilp32/libgcc'
make[3]: *** [Makefile:1201: multi-do] Error 1
make[3]: Leaving directory '/home/root/khan/riscv-gcc/build/riscv64-unknown-linux-gnu/libgcc'
make[2]: *** [Makefile:125: all-multi] Error 2
make[2]: Leaving directory '/home/root/khan/riscv-gcc/build/riscv64-unknown-linux-gnu/libgcc'
make[1]: *** [Makefile:12023: all-target-libgcc] Error 2
make[1]: Leaving directory '/home/root/khan/riscv-gcc/build'
make: *** [Makefile:887: all] Error 2

**From my point of view (or) If am not wrong, target board won't support 32bit, we need to build only 64bit Tool, Is there any possibility of building only 64bit Toolchain ?? (or) any suggestions**

kito-cheng, 2020-06-30:
Add `--disable-multilib` to configure option could disable that.

ghalib210, 2020-07-01:
Thanks a lot cheng, it worked.

closed 2020-07-01 by ghalib210