riscv-collab/riscv-gnu-toolchain #664: fatal errors: cannot find headers like <stdio.h>

cao1181320436, 2020-07-02:
As title described, when I was compiling my hello.c file, it showed up this error, and I cannot figure out what's going on.
Does anybody else have the same question as I do? Please help me. 
The whole error is as follows:

[image]

jim-wilson, 2020-07-02:
I would guess that your toolchain build did not complete, and you missed the fact that there was an error in the build.  With a correct complete build, the stdio.h file should be in /opt/riscvlinux/sysroot/usr/include/stdio.h.  There are also other ways that this could be failing though.  You should check to see if the stdio.h file exists in your install tree.  You can also try adding the -v option to the gcc command which will give verbose output, including the list of directories checked for header files, and you should verify that the dir that should contain the stdio.h file is on that list.  if the build did fail, and you have the build tree and build log, you should check for "error:" in the build log.  If you have the build tree but not the build log, you can just rerun the build, and it will rm -rf the last submodule build tree and try to rebuild it again and likely hit the error again.

kito-cheng, 2020-10-12:
Close this issue, seems issue resolved and long time no further discussion/response, feel free to create issue or reopen this issue if you have further question.

closed 2020-10-12 by kito-cheng