riscv-collab/riscv-gnu-toolchain #665: Regression for naked functions

fwsGonzo, 2020-07-02:
Hey guys,

After upgrading to the latest version mainly to test if performance got better and check out coroutines, I notice that my benchmarks simply didn't run anymore. A few hours later I noticed that if I disabled naked functions, everything started working again. I need the naked attribute to make the compiler stop adding prologue/epilogue code to entry functions. I'm using RISC-V as a script, so the functions become more or less like interrupt handlers.

As far as the details of the problem goes, it's hard to tell what exactly makes it happen. It's not completely broken, but it's also for sure the culprit because if I try to compile with -O1 and debugging it will just ICE like crazy.

```
[100%] Linking CXX executable rvbinary
/home/gonzo/github/script_bench/rvprogram/src/rvmain.cpp: In function 'selftest':
/home/gonzo/github/script_bench/rvprogram/src/rvmain.cpp:91:69: error: cannot allocate stack for variable '<anonymous>', naked function
   91 |  auto thread = microthread::create(function_that_adds, 111, 222, 333);
      |                                                                     ^
/home/gonzo/github/script_bench/rvprogram/src/rvmain.cpp:91:69: error: cannot allocate stack for variable '<anonymous>', naked function
/home/gonzo/github/script_bench/rvprogram/src/rvmain.cpp:91:69: error: cannot allocate stack for variable '<anonymous>', naked function
/home/gonzo/github/script_bench/rvprogram/src/rvmain.cpp:91:7: error: cannot allocate stack for variable 'thread', naked function
   91 |  auto thread = microthread::create(function_that_adds, 111, 222, 333);
```
So, I'm just writing all this while it's fresh in my memory but I haven't found a way to fully replicate this yet. I really like being able to used the naked attribute, even though its not 100% a necessity. And I really dislike going back in versions, and would like to stay on the latest version. So I will try to help figuring this out.

A minor related problem: In other languages like Rust that use LLVM as a backend the naked functions actually work better than on GCC because the return statement works. I don't know if GCC is just broken or if it's just something that nobody cares about, but return statements are just outright ignored. Thankfully I have many ways to make the sandbox stop running, including a custom instruction, so this doesn't affect me at all. It's just an observation.

-gonzo

jim-wilson, 2020-07-02:
On Thu, Jul 2, 2020 at 1:07 PM Alf-André Walla <notifications@github.com>
wrote:

> After upgrading to the latest version mainly to test if performance got
> better and check out coroutines, I notice that my benchmarks simply didn't
> run anymore. A few hours later I noticed that if I disabled naked
> functions, everything started working again. I need the naked attribute to
> make the compiler stop adding prologue/epilogue code to entry functions.
> I'm using RISC-V as a script, so the functions become more or less like
> interrupt handlers.
>
> /home/gonzo/github/script_bench/rvprogram/src/rvmain.cpp: In function 'selftest':
> /home/gonzo/github/script_bench/rvprogram/src/rvmain.cpp:91:69: error: cannot allocate stack for variable '<anonymous>', naked function
>    91 |  auto thread = microthread::create(function_that_adds, 111, 222, 333);
>       |                                                                     ^
>
>
The error looks correct from my point of view.  A naked function has no
prologue/epilogue, which means it has no frame.  And if you don't have a
frame, you can't allocate local variables on the stack.  it looks like a
bug that this error isn't generated always.  A little experimenting, and
for a simple testcase I get the error with -O1 but not -O2 with both gcc-9
and gcc-10.  So I think something is going wrong there preventing the error
from being reported.

In your case, there might have been optimizer heuristic change that caused
the code to work before but fail now, e.g. maybe something was inlined
before and isn't inline now because a heuristic changed, , or vice versa,
and whether the inlining happens affects whether we need a local variable
on the stack or not.

The return statement seems to work for simple examples.  But in general it
sounds like you are expecting too much from naked functions.  There is very
little that you can safely do in a naked function.

Jim

fwsGonzo, 2020-07-02:
I see, thanks for the detailed explanation.. I realized that I could continue using my fast instruction as a custom return statement so I'm not losing as much performance as I thought. I also want my environment to be rock solid, so I am probably not going to use naked any longer.

Do you know if there is any way to tell the compiler to stop saving/restoring registers for my entry functions? This is an emulated environment so each instruction is relatively expensive. Setting up the stack frame is completely fine, of course!

fwsGonzo, 2020-07-04:
Either way, as this is clearly not a regression I will just close this. Thanks again.

closed 2020-07-04 by fwsGonzo