riscv-collab/riscv-gnu-toolchain #667: checking for suffix of object files... configure: error: in `somepath/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d': configure: error: cannot compute suffix of object files: cannot compile

Anonymous-Someone, 2020-07-06:
I try to run 'make linux' but 1 error occur. My gcc and g++ version is from devtoolset-7  7.3.1
[image]

The build defaults to targetting RV64GC (64-bit), even on a 32-bit build environment. To build the 32-bit RV32GC toolchain, use:

**./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
make linux**

There is the config.log

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure --prefix=somepath/opt/riscv --with-arch=rv32gc --with-abi=ilp32d

## --------- ##
## Platform. ##
## --------- ##

hostname = eta.eecs.uci.edu
uname -m = x86_64
uname -r = 3.10.0-1127.10.1.el7.x86_64
uname -s = Linux
uname -v = #1 SMP Wed Jun 3 14:28:03 UTC 2020

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /opt/rh/devtoolset-7/root/usr/bin
PATH: /usr/lib64/qt-3.3/bin
PATH: /usr/local/bin
PATH: /usr/bin
PATH: /usr/local/sbin
PATH: /usr/sbin
PATH: .
PATH: /bin
PATH: /usr/bin
PATH: /usr/X11R6/bin
PATH: /usr/local/bin
PATH: /sbin
PATH: /usr/sbin
PATH: /usr/local/sbin
PATH: /opt/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1926: checking for gcc
configure:1942: found /opt/rh/devtoolset-7/root/usr/bin/gcc
configure:1953: result: gcc
configure:2182: checking for C compiler version
configure:2191: gcc --version >&5
gcc (GCC) 7.3.1 20180303 (Red Hat 7.3.1-5)
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2202: $? = 0
[... 213 lines cut ...]
with_gcc_src='$(srcdir)/riscv-gcc'
with_gdb_src='$(srcdir)/riscv-gdb'
with_glibc_src='$(srcdir)/riscv-glibc'
with_guile=''
with_linux_headers_src='$(srcdir)/linux-headers/include'
with_musl_src='$(srcdir)/riscv-musl'
with_newlib_src='$(srcdir)/riscv-newlib'
with_system_zlib='--with-system-zlib'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
#define HAVE_LIBGMP 1
#define HAVE_LIBMPFR 1
#define HAVE_LIBMPC 1

configure: exit 0

kito-cheng, 2020-07-06:
Could you provide config.log from `build-glibc-linux-rv32gc-ilp32d`? the config.log in top-level build folder didn't contain too much information for debugging.

Anonymous-Someone, 2020-07-06:
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by GNU C Library configure (see version.h), which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ somepath/riscv-gnu-toolchain/riscv-glibc/configure --host=riscv32-unknown-linux-gnu --prefix=/usr --disable-werror --enable-shared --enable-obsolete-rpc --with-headers=/data/home/xiaosug/riscv-gnu-toolchain/linux-headers/include --disable-multilib --enable-kernel=3.0.0 --libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib

## --------- ##
## Platform. ##
## --------- ##

hostname = eta.eecs.uci.edu
uname -m = x86_64
uname -r = 3.10.0-1127.10.1.el7.x86_64
uname -s = Linux
uname -v = #1 SMP Wed Jun 3 14:28:03 UTC 2020

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /data/home/xiaosug/riscv-gnu-toolchain/scripts/wrapper/sed
PATH: /data/home/xiaosug/riscv-gnu-toolchain/../opt/riscv/bin
PATH: /opt/rh/devtoolset-7/root/usr/bin
PATH: /usr/lib64/qt-3.3/bin
PATH: /usr/local/bin
PATH: /usr/bin
PATH: /usr/local/sbin
PATH: /usr/sbin
PATH: .
PATH: /bin
PATH: /usr/bin
PATH: /usr/X11R6/bin
PATH: /usr/local/bin
PATH: /sbin
PATH: /usr/sbin
PATH: /usr/local/sbin
PATH: /opt/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2218: checking build system type
configure:2232: result: x86_64-pc-linux-gnu
configure:2252: checking host system type
configure:2265: result: riscv32-unknown-linux-gnu
configure:2294: checking for riscv32-unknown-linux-gnu-gcc
configure:2321: result: riscv32-unknown-linux-gnu-gcc 
configure:2590: checking for C compiler version
configure:2599: riscv32-unknown-linux-gnu-gcc  --version >&5
/data/home/xiaosug/riscv-gnu-toolchain/riscv-glibc/configure: line 2601: riscv32-unknown-linux-gnu-gcc: command not found
configure:2610: $? = 127
configure:2599: riscv32-unknown-linux-gnu-gcc  -v >&5
/data/home/xiaosug/riscv-gnu-toolchain/riscv-glibc/configure: line 2601: riscv32-unknown-linux-gnu-gcc: command not found
configure:2610: $? = 127
configure:2599: riscv32-unknown-linux-gnu-gcc  -V >&5
/data/home/xiaosug/riscv-gnu-toolchain/riscv-glibc/configure: line 2601: riscv32-unknown-linux-gnu-gcc: command not found
configure:2610: $? = 127
configure:2599: riscv32-unknown-linux-gnu-gcc  -qversion >&5
/data/home/xiaosug/riscv-gnu-toolchain/riscv-glibc/configure: line 2601: riscv32-unknown-linux-gnu-gcc: command not found
configure:2610: $? = 127
[... 222 lines cut ...]
subdirs=' '
submachine=''
sysconfdir='${prefix}/etc'
sysheaders=''
sysnames=''
target_alias=''
use_default_link=''
use_ldconfig=''
use_nscd=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "GNU C Library"
#define PACKAGE_TARNAME "glibc"
#define PACKAGE_VERSION "(see version.h)"
#define PACKAGE_STRING "GNU C Library (see version.h)"
#define PACKAGE_BUGREPORT "https://sourceware.org/bugzilla/"
#define PACKAGE_URL "http://www.gnu.org/software/glibc/"
#define PKGVERSION "(GNU libc) "
#define REPORT_BUGS_TO "<http://www.gnu.org/software/libc/bugs.html>"

configure: exit 1

kito-cheng, 2020-07-06:
Seems like the root cause is ` riscv32-unknown-linux-gnu-gcc: command not found`, that mean gcc is build failed, could you try  `rm stamps/build-gcc-linux-stage1` and `make linux` again?

Anonymous-Someone, 2020-07-06:
It does not change.
[image]
And I run other commands:
./configure --prefix=/opt/riscv --enable-multilib
make linux
there are 2 errors
[image]

kito-cheng, 2020-07-06:
Could you did a clean build in another folder? reconfigure in same folder would be error prone.
e.g.
```
$ mkdir build-riscv-gnu-toolchain
$ cd build-riscv-gnu-toolchain
$ <PATH-TO-RISCV-GNU-TOOLCHAIN>/configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
make linux
```

jim-wilson, 2020-07-06:
Builds work best if the --prefix dir is empty.  If you already have a build in $prefix, and it was configured differently than what you are trying to build now, the build can fail in odd ways.  This is mentioned in the trouble shooting section of the README.md file.  Personally, I would suggest always using a different $prefix dir for every build.

From the configure logs, it looks like riscv32-unknown-elf-gcc was found, but was not able to be run.  This can sometimes happen if you have a binary which was linked against a dynamic linker version that doesn't exist anymore.  Try running ldd on the riscv32-unknown-elf-gcc binary to check that.  If you get errors you have an old useless binary, maybe one that was built on an older OS version, or built for a different linux distro.

cypox, 2022-03-10:
`make clean` fixed it for me. I had already built the 64 suite and tried to re-configure and build the 32-version. You should clean before. You will keep your binaries if you dont want to build twice.

TommyMurphyTM1234, 2022-09-27:
Resolved.

closed 2022-09-27 by TommyMurphyTM1234