riscv-collab/riscv-gnu-toolchain #668: Execute-only .text section

fwsGonzo, 2020-07-06:
Would it be possible to have execute-only code as a feature... somewhere in the toolchain? How would that work and which program would be responsible for supporting such a feature? Looking at readelf the tool doesn't really tell you if a section is readable or not - even though the ELF format does have a bit for that permission.

What do you think are the pros and cons of this feature? Afaik RISC-V doesn't really need to read the .text segment outside of executing the machine code. Maybe libunwind needs to read it, or is the .eh_frame information rich enough?

You can find some light information here:
https://linuxplumbersconf.org/event/4/contributions/283/attachments/357/588/Touch_but_dont_look__Running_the_kernel_in_execute_only_memory-presented.pdf

-gonzo

fwsGonzo, 2020-09-25:
Are linker scripts too inflexible for this?

jim-wilson, 2020-09-25:
readelf -l will show you program headers, and there are flag bits RWE for read write and execute.  Normally read-only data is put in the text segment, that would have to change.  A debugger needs read access to code if you want to show disassembly of the executing code.  libunwind doesn't read code, it only uses the unwind info.  Otherwise, you would have to try it and see what breaks.

jim-wilson, 2020-09-25:
you should be able to do this by modifying the default linker script

fwsGonzo, 2020-09-26:
Thanks, I am in the process of doing that now. I am having problems isolating text from rodata though, and while the linker accepts my NOREAD attribute on .text, it doesn't seem to honor it:

```
	.text    NOREAD :
	{
	  *(.text.unlikely .text.*_unlikely .text.unlikely.*)
	  *(.text.exit .text.exit.*)
	  *(.text.startup .text.startup.*)
	  *(.text.hot .text.hot.*)
	  *(SORT(.text.sorted.*))
	  *(.text .stub .text.* .gnu.linkonce.t.*)
	}
	. = ALIGN(0x1000);

  /* Read-only sections, merged into text segment: */
  .init           :
  {
    KEEP (*(SORT_NONE(.init)))
  }
```
--->
```
Program Headers:
  Type           Offset             VirtAddr           PhysAddr
                 FileSiz            MemSiz              Flags  Align
  LOAD           0x0000000000000000 0x000000000011f000 0x000000000011f000
                 0x0000000000030158 0x0000000000030158  R E    0x1000
  LOAD           0x0000000000030158 0x0000000000150158 0x0000000000150158
                 0x0000000000008992 0x0000000000009290  RW     0x1000
```

fwsGonzo, 2020-09-26:
Having investigated this some more, here are my conclusions:
linker scripts don't support setting permissions, leaving only the linker to separate executable code and rodata.
ld does not support execute-only segments, nor does it support separating .text and .rodata.

However, gold supports it. `man gold` shows:
```
       --rosegment
              Put read-only non-executable sections in their own segment
```
I'm guessing that gold does not support elf64lriscv, though. So I am out of luck.
I did notice that `ld` does support custom options for specific emulations, and so it should be possible to add a -z option for RISC-V that does separate the two segments, even if it does not change segment permissions.

What do you think?

fwsGonzo, 2020-09-26:
I tested this method in my emulator. I just ditched the execute pages completely and ran instructions off an array of instruction function pointers. The programs are running as expected. Execute-only works on RISC-V.

There are other challenges, like I am still not able to remove the read-bit from execute segment.
I had to modify the linker script a bit as it's placing lots of data around .text, such as .init, which is just an array of function pointers, and is usually well inside .rodata.

I'll keep this open until I can figure out a way to get rid of the read bit on my .text section. The reason why is because I can't differentiate between the segments. I have code that depends on rodata being linear. If rodata and text is merged, that's fine. But if they aren't how can I detect it when text is RX? I can't, unless I guess that RX really means X because there are 3 segments. I need a way to detect if the text segment is really execute-only. Otherwise I am writing broken code.

fwsGonzo, 2020-09-27:
The linker script I am using now. It's not much, but it works:
https://gist.github.com/fwsGonzo/9184d9352ef37bce0ad03e4cef0c0f13

To remove the read-bit from the execute segment I used this chperm code I found. It had some bugs and annoyances, but they are largely fixed now. It really only needs to support 32-bit ELFs as well:
https://gist.github.com/fwsGonzo/e3dbcd9097953e7e113fd3f3ada04316

closed 2020-09-27 by fwsGonzo

jim-wilson, 2020-09-27:
You might try discussing this on the FSF Binutils mailing list.  The fundamental issues are not RISC-V specific, and the vast majority of binutils developers won't see anything posted here.