riscv-collab/riscv-gnu-toolchain #669: ld: cannot find -lstdc++ / -lgcc_s

wky9710, 2020-07-07:
I was building chipyard and BOOM, (instructions here: https://docs.boom-core.org/en/latest/)
building riscv toolchain, like this
`./scripts/build-toolchains.sh riscv-tools`
and errors below:
```
configure: WARNING: gcc emits a warning for alias between functions of incompatible types
configure: WARNING: Multi-arch is disabled.
configure: WARNING: gcc emits a warning for alias between functions of incompatible types
configure: WARNING: Multi-arch is disabled.
configure: WARNING: minimum kernel version reset to 4.15.0
configure: WARNING: minimum kernel version reset to 4.15.0
*** errlist.c count 134 inflated to GLIBC_2.12 count 135 (old errno.h?)
/home/wangkunyu/work/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lstdc++
/home/wangkunyu/work/chipyard/riscv-tools-install/lib/gcc/riscv64-unknown-linux-gnu/9.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lgcc_s
collect2: error: ld returned 1 exit status
make[3]: *** [/home/wangkunyu/work/chipyard/toolchains/riscv-tools/riscv-gnu-toolchain/build/build-glibc-linux-rv64imafdc-lp64d/support/links-dso-program] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [support/others] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-glibc-linux-rv64imafdc-lp64d] Error 2
```
seems ld in riscv-tools failed
but ld on my lab's server _can_ find lstdc++ / lgcc_s, details below
why ld in riscv tools can't find these two libs? the version or as #498 mentioned, --libdir? (I can't find where to set --libdir...)
p.s., lstdc++ and lgcc_s are gcc version 9.2.0, located in /lib

I'm really new to this area, thanks for your patience :)

```
wangkunyu@server17:~/work/chipyard/riscv-tools-install/riscv64-unknown-linux-gnu/bin$ ld -lstdc++
ld: warning: cannot find entry symbol _start; not setting start address
wangkunyu@server17:~/work/chipyard/riscv-tools-install/riscv64-unknown-linux-gnu/bin$ ld --version
GNU ld (GNU Binutils for Ubuntu) 2.26.1
Copyright (C) 2015 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
wangkunyu@server17:~/work/chipyard/riscv-tools-install/riscv64-unknown-linux-gnu/bin$ ./ld -lstdc++
./ld: cannot find -lstdc++
wangkunyu@server17:~/work/chipyard/riscv-tools-install/riscv64-unknown-linux-gnu/bin$ ./ld --version
GNU ld (GNU Binutils) 2.32
Copyright (C) 2019 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
wangkunyu@server17:~/work/chipyard/riscv-tools-install/riscv64-unknown-linux-gnu/bin$ ld -lgcc_s
ld: warning: cannot find entry symbol _start; not setting start address
wangkunyu@server17:~/work/chipyard/riscv-tools-install/riscv64-unknown-linux-gnu/bin$ ./ld -lgcc_s
./ld: cannot find -lgcc_s

```

lbmeng, 2020-09-24:
I suspect in your $PATH there is already a `riscv64-unknown-linux-gnu-gcc` which confused the build process.

lazyparser, 2021-03-04:
@wky9710 Hi, have you fixed it?

wky9710, 2021-03-05:
I changed a server(because it was on my lab's server with version 2.26.1, so i can't upgrade it orz)
ld version: 2.26.1 to 2.33.1
though it still says "cannot find -lstdc++ / -lgcc_s", but the build succeeds...
Seems it's the ld version's problem... In all it works orz

closed 2021-03-05 by lazyparser

dovedevic, 2021-03-22:
I'm running into this issue as well, invoking a compile command from a SPEC benchmark as follows:

```bash
$ /home/dovedevic/local/riscvv09/gnu/bin/riscv64-unknown-elf-g++     -march=rv64gcv      -g -O3  stopwatch.o FullBoard.o KoState.o Playout.o TimeControl.o UCTSearch.o GameState.o Leela.o SGFParser.o Timing.o Utils.o FastBoard.o Matcher.o SGFTree.o TTable.o Zobrist.o FastState.o GTP.o MCOTable.o Random.o SMP.o UCTNode.o    -o leela_r
/home/dovedevic/local/riscvv09/gnu/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: cannot find -lstdc++
```

When inspecting the `/home/dovedevic/local/riscvv09/gnu/lib/gcc/riscv64-unknown-elf/9.2.0/` directory, I have the following: 

`crtbegin.o, crtend.o, crti.o, crtn.o, /include, /include-fixed, /install-tools, libgcc.a, libgcov.a, /plugin, /rv32i, /rv32iac, /rv32im, /rv32imac, /rv32imafc, /rv64imac, /rv64imafdc`

Should I be seeing a `libg++.a` here? Or should there be an entire other directory `/home/dovedevic/local/riscvv09/gnu/lib/g++` in the install dir, and if so, where do I specify that in the install/build to do so?

Or should I be specifying the -ldir to something else specific. Any guidance is much appreciated.

jim-wilson, 2021-03-22:
This rvv library problem is likely unrelated to the original problem.  There are many different ways you can hit this problem.

You can find the location of libstdc++.a by adding --print-file-name=llibstdc++.a to the gcc command.  If it just prints libstdc++.a instead of a full path then you don't have one.  Note that the -march and -mabi options are significant here.  Try it with and without the -march=rv64gcv option.

I see that you are asking for vector code with -march=rv64gcv.  Did you build with rv64gcv as the default arch?  Did you build multilib?  If you built multilib, did you build a rv64gcv multilib?  You can see the list of multilibs by running gcc and adding the --print-multi-lib option.  It looks like you built multilib, and you didn't build a rv64gcv multilib, because there is no dir for rv64imafdcv.

The primary focus for vector work is LLVM at the moment, and the gcc vector work is not actively being developed.  I don't think that we added a multilib for vector support.  It probably only works right if you configure rv64gcv as the default arch.

dovedevic, 2021-03-22:
@jim-wilson I checked the libstdc++ and multilib file paths and they all do exist, however I did not contain a lib for the rv64imafdcv, and only the rv64imafdc. Since in my workload I am not using the vector extension yet, I omitted it in the march specification to just rv64gc and all is well. 

I'll look into getting a multilib built for the rv64imafdcv when the time is needed. Many thanks.