riscv-collab/riscv-gnu-toolchain #670: riscv64-unknown-linux-gnu-gcc: error: unrecognized command-line option '-m64'

rmuneeswaran, 2020-07-10:
I porting the openssl for arch riscv but when I did make the riscv compiler throwing -m64 option was not unrecognised.  I would like to understand the m64 option supported in the gcc 10.1.0 version.

openssl/openssl-1.1.1g$ ./config --prefix=$PWD/riscv --cross-compile-prefix=riscv64-unknown-linux-gnu-

and make failed with unrecognised options for -m64. whether I am missing any configuration support for -m64 option. Please share your inputs.

~/openssl/openssl-1.1.1g$ make
/usr/bin/perl "-I." -Mconfigdata "util/dofile.pl" \
    "-oMakefile" include/crypto/bn_conf.h.in > include/crypto/bn_conf.h
/usr/bin/perl "-I." -Mconfigdata "util/dofile.pl" \
    "-oMakefile" include/crypto/dso_conf.h.in > include/crypto/dso_conf.h
/usr/bin/perl "-I." -Mconfigdata "util/dofile.pl" \
    "-oMakefile" include/openssl/opensslconf.h.in > include/openssl/opensslconf.h
make depend && make _all
make[1]: Entering directory '/home/span48/openssl/openssl-1.1.1g'
make[1]: Leaving directory '/home/span48/openssl/openssl-1.1.1g'
make[1]: Entering directory '/home/span48/openssl/openssl-1.1.1g'
riscv64-unknown-linux-gnu-gcc  -I. -Iinclude -fPIC -pthread -m64 -Wa,--noexecstack -Wall -O3 -DOPENSSL_USE_NODELETE -DL_ENDIAN -DOPENSSL_PIC -DOPENSSL_CPUID_OBJ -DOPENSSL_IA32_SSE2 -DOPENSSL_BN_ASM_MONT -DOPENSSL_BN_ASM_MONT5 -DOPENSSL_BN_ASM_GF2m -DSHA1_ASM -DSHA256_ASM -DSHA512_ASM -DKECCAK1600_ASM -DRC4_ASM -DMD5_ASM -DAESNI_ASM -DVPAES_ASM -DGHASH_ASM -DECP_NISTZ256_ASM -DX25519_ASM -DPOLY1305_ASM -DOPENSSLDIR="\"/home/span48/openssl/openssl-1.1.1g/riscv/ssl\"" -DENGINESDIR="\"/home/span48/openssl/openssl-1.1.1g/riscv/lib/engines-1.1\"" -DNDEBUG  -MMD -MF apps/app_rand.d.tmp -MT apps/app_rand.o -c -o apps/app_rand.o apps/app_rand.c
riscv64-unknown-linux-gnu-gcc: error: unrecognized command-line option '-m64'
Makefile:708: recipe for target 'apps/app_rand.o' failed
make[1]: *** [apps/app_rand.o] Error 1
make[1]: Leaving directory '/home/span48/openssl/openssl-1.1.1g'
Makefile:174: recipe for target 'all' failed
make: *** [all] Error 2

I have given gcc spec  details as below.

 riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-linux-gnu/10.1.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: $HOME/riscv/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.1.0 (GCC)

jim-wilson, 2020-07-10:
There is no -m64 option.  A riscv64-unknown-linux-gnu-gcc compiler emits 64-bit code by default, so there is no need for any option here.

We don't support -m64 because we have more than one 64-bit choice.  There is a -march=X option that accepts architecture names as specified via the unprivileged spec, e.g. -march=rv64imac (no fp) or -march=rv64imafdc (with fp).  There is a -mabi=X option that accepts ABI choices as specified in the psABI, e.g. -mabi=lp64 (soft-float) -mabi=lp64d (hard-float).  There are also 32-bit choices for arch and abi.

rmuneeswaran, 2020-07-13:
Thanks for more details.

TommyMurphyTM1234, 2022-09-27:
Invalid.