riscv-collab/riscv-gnu-toolchain #673: there was some errors, like 'bfd.texi:193: @include: could not find bfdsumm.texi'.

Martion-z, 2020-07-17:
....
creating bfdver.texi
restore=: && backupdir=".am$$" && \
rm -rf $backupdir && mkdir $backupdir && \
if (makeinfo --split-size=5000000 --split-size=5000000 --version) >/dev/null 2>&1; then \
  for f in bfd.info bfd.info-[0-9] bfd.info-[0-9][0-9] bfd.i[0-9] bfd.i[0-9][0-9]; do \
    if test -f $f; then mv $f $backupdir; restore=mv; else :; fi; \
  done; \
else :; fi && \
if makeinfo --split-size=5000000 --split-size=5000000   -I /home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/riscv-binutils/bfd/doc \
 -o bfd.info `test -f 'bfd.texi' || echo '/home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/riscv-binutils/bfd/doc/'`bfd.texi; \
then \
  rc=0; \
else \
  rc=$?; \
  $restore $backupdir/* `echo "./bfd.info" | sed 's|[^/]*$||'`; \
fi; \
rm -rf $backupdir; exit $rc
/home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/riscv-binutils/bfd/doc/bfd.texi:193: @include: could not find bfdsumm.texi
/home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/riscv-binutils/bfd/doc/bfd.texi:318: @include: could not find fdl.texi
make[4]: *** [Makefile:541：bfd.info] 错误 1
make[4]: 离开目录“/home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/build-binutils-newlib/bfd/doc”
make[3]: *** [Makefile:1641：info-recursive] 错误 1
make[3]: 离开目录“/home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/build-binutils-newlib/bfd”
make[2]: *** [Makefile:2690：all-bfd] 错误 2
make[2]: 离开目录“/home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/build-binutils-newlib”
make[1]: *** [Makefile:849：all] 错误 2
make[1]: 离开目录“/home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/build-binutils-newlib”
make: *** [Makefile:391：stamps/build-binutils-newlib] 错误 2

jim-wilson, 2020-07-17:
The makeinfo command has "-I /home/zxjbeauty/桌面/riscv/riscvtc/riscv-gnu-toolchain/riscv-binutils/bfd/doc" and the files should be in that directory.  Did you check to make sure that the files are there?  If the files are there, then maybe it is a problem with the non-ascii chars in the pathname confusing makeinfo.  Maybe you can try using a pathname with only ascii characters.  Since makeinfo did find the files in the build directory, you could try copying the files you need from the source dir to the build dir as a workaround.  The makefile will rm -rf the build dir on a rebuild though, so you would have to cd into the binutils build dir, make, install, and then touch the appropriate stamp file.  Or alternatively, modify the makefile to copy the two files into the build dir before the build.

Martion-z, 2020-07-18:
Thanks! That's exactly the error caused by non-ascii characters! Since I change a pathname with only ascii characters,  successfully maked.

SineStriker, 2022-07-06:
Thanks, I encountered this problem too.

TommyMurphyTM1234, 2022-09-27:
Resolved.

closed 2022-09-27 by TommyMurphyTM1234