riscv-collab/riscv-gnu-toolchain #675: Unable to produce rv64g linux build environment

fwsGonzo, 2020-07-25:
```
./configure --prefix=/home/gonzo/riscv --with-arch=rv64g --with-abi=lp64d --enable-multilib
make linux
```
->
```
$ riscv64-unknown-linux-gnu-gcc --print-multi-lib
lib32/ilp32;@march=rv32imac@mabi=ilp32
lib32/ilp32d;@march=rv32imafdc@mabi=ilp32d
lib64/lp64;@march=rv64imac@mabi=lp64
lib64/lp64d;@march=rv64imafdc@mabi=lp64d
```
Everything contains compressed instructions. What's the reason it happens?

jim-wilson, 2020-07-26:
--enable-multilib substitutes a default list of arch/abi choices, none of which are the one you specified.  You could try modifying the sources to make the default list of multilibs include the one you want.  But the simpler solution is to just not use --enable-multilib if you want a specific arch/abi choice that isn't one of the defaults.

fwsGonzo, 2020-07-26:
Exactly as you said. Worked like a charm.

closed 2020-07-26 by fwsGonzo