riscv-collab/riscv-gnu-toolchain #677: build from source fails on mac os mojave

adityaatluri, 2020-07-28:
Hi,
I did
```
./configure --prefix=$INSTALL_DIR
make linux
```
system:
MacOS Mojave
```
$ cc --version
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

Got the following error:
```
Checking multilib configuration for libgcc...
mkdir riscv64-unknown-linux-gnu
mkdir riscv64-unknown-linux-gnu/libgcc
Configuring in riscv64-unknown-linux-gnu/libgcc
configure: creating cache ./config.cache
checking build system type... x86_64-apple-darwin18.7.0
checking host system type... riscv64-unknown-linux-gnu
checking for --enable-version-specific-runtime-libs... no
checking for a BSD-compatible install... /usr/bin/install -c
checking for gawk... /usr/local/bin/gawk
checking for riscv64-unknown-linux-gnu-ar... /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/bin/ar
checking for riscv64-unknown-linux-gnu-lipo... riscv64-unknown-linux-gnu-lipo
checking for riscv64-unknown-linux-gnu-nm... /Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/nm
checking for riscv64-unknown-linux-gnu-ranlib... /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/bin/ranlib
checking for riscv64-unknown-linux-gnu-strip... /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/bin/strip
checking whether ln -s works... yes
checking for riscv64-unknown-linux-gnu-gcc...  /Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/bin/ -B/Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/lib/ -isystem /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/include -isystem /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/sys-include   
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether  /Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/bin/ -B/Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/lib/ -isystem /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/include -isystem /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/sys-include    accepts -g... yes
checking for  /Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/bin/ -B/Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/lib/ -isystem /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/include -isystem /Users/aditya/.local/riscv-gnu/riscv64-unknown-linux-gnu/sys-include    option to accept ISO C89... unsupported
checking how to run the C preprocessor... /lib/cpp
configure: error: in `/Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc':
configure: error: C preprocessor "/lib/cpp" fails sanity check
See `config.log' for more details
make[1]: *** [Makefile:12885: configure-target-libgcc] Error 1
make[1]: Leaving directory '/Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:276: stamps/build-gcc-linux-stage1] Error 2
```

jim-wilson, 2020-07-28:
Building on a mac is tricky.  You need to use homebrew to install a lot of GNU packages, and then you have to define some Makefile variables to force the build to use the homebrew version of some tools instead of the mac version, e.g. AWK=gawk.

Also, I don't know if it is possible to build a linux toolchain on a mac.  The linux toolchain build requires a case sensitive file system.  I think macs use a case insensitive filesystem.  Unless maybe there is some way that you can arrange that a disk partition is treated as case sensitive, and then do the build inside that disk partition.

I don't have access to a mac, so I don't know the details.

sifive/freedom-tools has support to build on a mac and shows what is necessary to make it work.  But this only builds an embedded cross compiler.  It won't build a linux cross compiler on a mac.

adityaatluri, 2020-07-28:
Right. I have all the dependency packages installed. I am curious why it builds fine on Catalina but not on Mojave. Do you know what the error is about? `config.log` file is 
```
configure:1926: checking for gcc
configure:1942: found /usr/bin/gcc
configure:1953: result: gcc
configure:2182: checking for C compiler version
configure:2191: gcc --version >&5
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
configure:2202: $? = 0
configure:2191: gcc -v >&5
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin18.7.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
configure:2202: $? = 0
configure:2191: gcc -V >&5
clang: error: unsupported option '-V -Wno-framework-include-private-from-public'
clang: error: no input files
configure:2202: $? = 1
configure:2191: gcc -qversion >&5
clang: error: unknown argument '-qversion', did you mean '--version'?
clang: error: no input files
```

Looks like I may have to install regular gnu toolchain (x86 + darwin) to resolve this error?

jim-wilson, 2020-07-28:
Different compilers from different vendors support different command line options, so configure just tries them all to see which ones work with the compiler you are using.  This stuff is normal in configure output and is not an error.

There are multiple config.log files, one in each directory that has a configure script, so you need to make sure you are looking at the right one.

The first message mentions
checking how to run the C preprocessor... /lib/cpp
configure: error: in `/Volumes/riscv/src/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc':
configure: error: C preprocessor "/lib/cpp" fails sanity check
See `config.log' for more details

so it looks like a problem with /lib/cpp.  And this points at the libgcc build dir that contains the config.log file that you need to look at to see the /lib/cpp error.

adityaatluri, 2020-07-28:
This is what I see in config.log
```
/Volumes/riscv/src/riscv-gnu-toolchain/riscv-gcc/libgcc/configure: line 1527: /lib/cpp: No such     file or directory
```

jim-wilson, 2020-07-28:
If you look at the configure script, and search for "checking how to run the C preprocessor" you will see that it tries three things in series, and then fails only if all 3 fail
    for CPP in "$CC -E" "$CC -E -traditional-cpp" "/lib/cpp"
if $CC is gcc then one of the first two should have worked.  I don't know about clang.  If config.log, you should see output for trying all 3 possibilities.  Check to see why the failed.  One of them probably should not have failed.  Linux systems are required to have /lib/cpp by the FHS so that is a reasonable last chance thing to try, but I don't know about mac os.

adityaatluri, 2020-08-03:
Switched to Catalina so that binaries are installed instead of building from source. Working fine on catalina using homebrew

closed 2020-08-03 by adityaatluri