riscv-collab/riscv-gnu-toolchain #680: Why asembler `uint32_t j = 4294967293u;` to `li a5,-2`

luyahan, 2020-07-30:
I comlie a program by riscv-g++.  I find `uint32_t j = 4294967293u` is complied to `li a5,-3` not `li a5,4294967293`(a5 is `0xfffffffffffffffd`). why do it?
In aarch64-gcc, `uint32_t j = 4294967293u;`  is complied to ` mov     w0, #0xfffffffd                 // #-3 `

luyahan, 2020-07-30:
```
uint32_t j = 2147483648u;
```
is compiled to 
```
   10642:       800007b7                lui     a5,0x80000
   10646:       fef42423                sw      a5,-24(s0)
   1064a:       fe842783                lw      a5,-24(s0)
```
the a5 is 0xffffffff80000000.Why do lw not lwu?

luyahan, 2020-07-30:
I am porting v8 to riscv64. I encounter o problem in a test case:
```
TEST(RunUint32MulHighP) {
  RawMachineAssemblerTester<int32_t> m;
  Int32BinopTester bt(&m);
  bt.AddReturn(m.Uint32MulHigh(bt.param0, bt.param1));
  uint32_t i = 4294967295, j = 4294967295;
  int32_t expected = bit_cast<int32_t>(static_cast<uint32_t>(
      (static_cast<uint64_t>(i) * static_cast<uint64_t>(j)) >> 32));
  std::cout << expected << "," << bt.call(bit_cast<int32_t>(i), bit_cast<int32_t>(j))
            << "," << i << "," << j << std::endl;
  std::cout << sizeof(expected) << "," << sizeof(bt.call(bit_cast<int32_t>(i), bit_cast<int32_t>(j))) << std::endl;
  int32_t result = bt.call(bit_cast<int32_t>(i), bit_cast<int32_t>(j));
  std::cout << "result:" << result << std::endl;
  CHECK_EQ(expected, bt.call(bit_cast<int32_t>(i), bit_cast<int32_t>(j)));
}
````
output:
````
-2,4294967294,4294967295,4294967295
4,4
result:4294967294
```
Uint32MulHigh set 0xfffffffd to a0. so result be seted 0xfffffffd  not be singed extended.

sorear, 2020-07-30:
The intent of the RISC-V ISA is that 32-bit C values are stored sign extended in registers, even for unsigned types.  So `li a5, -5` produces the correct in-memory representation of a `uint32_t`.  All 32-bit operations (addw, subw, addiw, mulw, etc) generate sign extended values, logical operations (and, or, xor, sll, ...) generate sign-extended results from sign-extended inputs, comparisons (including unsigned comparisons) produce the correct result if both arguments are sign-extended, and stores don't care either way about the upper half because they only store the lower half.

This does mean that widening casts from uint32_t to uint64_t are not no-ops; gcc generates them as two shift instructions (sll 32, srl 32).

In the specific case of lw versus lwu, lw is a compressible instruction and should be used unless there is a compelling need for lwu.

The behavior of the C++ code in your example looks correct to me; not familiar with v8 Uint32MulHigh.

luyahan, 2020-07-30:
> The intent of the RISC-V ISA is that 32-bit C values are stored sign extended in registers, even for unsigned types.  So `li a5, -5` produces the correct in-memory representation of a `uint32_t`. [...]

Thank you, i got it. I deal with uint without signed extened. It leads to the 'result' is not correct.

closed 2020-07-30 by luyahan

luyahan, 2020-09-03:
> uint32_t

If i want get a uint32_t with zero-extension, can riscv-gcc do it ?

aswaterman, 2020-09-03:
The sign-extension is an ABI constraint.  If you want zero extension, you need to use a wider unsigned type, like `uint64_t`.