riscv-collab/riscv-gnu-toolchain #681: About GDB. 0x0 in fs9, isn't fs9 (float = NaN)?

luyahan, 2020-07-31:
[image]
0x0 in fs9, is not fs9 (float = NaN)?
I think 0.0 in fs9 should be 0xffffffff00000000.
refer to https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/mfGw77uQYwc/m/1IrLgMpWAwAJ

luyahan, 2020-07-31:
My gdb version:
```
GNU gdb (GDB) Fedora 9.0.50.20191018-1.0.riscv64.fc32
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

jim-wilson, 2020-07-31:
I assume you are using qemu?  It is a known bug in qemu that it doesn't handle NaN-boxing correctly.  This does work on hardware.  There is no gdb problem here.  I believe that there are patches in progress for qemu to fix this.

(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/jimw/tmp/a.out 

Breakpoint 1, sub () at tmp2.c:3
3	  return 0.0;
1: x/i $pc
=> 0x10438 <sub+6>:	fmv.w.x	fa5,zero
(gdb) stepi
4	}
1: x/i $pc
=> 0x1043c <sub+10>:	fmv.s	fa0,fa5
(gdb) print $fa5
$2 = {float = 0, double = -nan(0xfffff00000000)}
(gdb)

luyahan, 2020-08-03:
Refer https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/mfGw77uQYwc/m/HhOb3QUvBAAJ

closed 2020-08-03 by luyahan