riscv-collab/riscv-gnu-toolchain #682: How to build a toolchain for rv32imafc target, but with 8bytes stack alignment?

hotislandn, 2020-08-03:
Thanks for your toolchian.
Since our target IC only has limited RAM, and default stack boundary for rv32imafc is 16bytes, accroding to RV32 spec.
We tried to put -mpreferred-stack-boundary=3 into current toolchain's options, but it fails to link to c lib.
We are now trying to build a rv32imafc toolchain from scratch via your code but with 8bytes stack boundary.
How can we achieve this?
Thanks in advance and have a nice day!

jim-wilson, 2020-08-03:
The precompiled C libraries were compiled with 16-byte alignment for the stack.  As long as you don't have rv128 or Q support, and don't use any 128-bit types like long double or __int128_t, and don't request 128-bit alignment via an attribute, then 8 byte stack support should work for user code, but you will need to build your own C libraries with the right stack alignment.  I don't know offhand if 8-byte alignment works for privileged code, but if necessary you can have a separate stack for that.

There are a few different ways to accomplish this.  You can add your desired option to MULTILIB_DEFAULTS if the gcc/config/riscv/t-elf-multilib file.  The compiler will then fail to link with libraries if you forget the option.  Or you can hard wire on the options in gcc/config/riscv/riscv.opt after Var(riscv_preferred_stack_boundary_arg) add Init(3) which is safer.  Or you can set the default stack boundary to 64-bits by changing ABI_STACK_BOUNDARY in gcc/config/riscv/riscv.h.  Or you can try adding the option to the multilib list so you get libraries compiled with 128-bit aligned stack and 64-bit aligned stack.  This is the most complicated option.  I don't think we have support for this in the multilib-generator script, so you probably have to extend the script or fix files by hand.  Maybe not worth the effort.  The second and third options are most likely to work so I'd try one of those.  You might need to experiment to get this working.

hotislandn, 2020-08-04:
Hi Jim,

Thanks for your detailed advice.
We have tried to hack the code as what you said:

--- a/gcc/config/riscv/riscv.opt
+++ b/gcc/config/riscv/riscv.opt
@@ -34,7 +34,7 @@ Target Report RejectNegative Joined Enum(abi_type) Var(riscv_abi) Init(ABI_ILP32
 Specify integer and floating-point calling convention.

 mpreferred-stack-boundary=
-Target RejectNegative Joined UInteger Var(riscv_preferred_stack_boundary_arg)
+Target RejectNegative Joined UInteger Var(riscv_preferred_stack_boundary_arg) **Init(3)**
 Attempt to keep stack aligned to this power of 2.

and re-compiled the toolchain successfully. 

We now tried our new toolchain for code compiling, and it shows that, the stack alignment boundary is 8 bytes now:

54000f18 <clic_init>:
                interrupt_function_ptr_t* vect_table,
                interrupt_function_ptr_t default_handler,
                uint32_t num_irq,
                uint32_t num_config_bits
                )
{
54000f18:	fd810113          	addi	sp,sp,**-40**

void default_handler(void)
{
54000ff8:	1161                	addi	sp,sp,**-8**
54000ffa:	c222                	sw	s0,4(sp)

the previous toolchain will allocate on 16bytes boundary here, just as:

54000f28 <clic_init>:
                interrupt_function_ptr_t* vect_table,
                interrupt_function_ptr_t default_handler,
                uint32_t num_irq,
                uint32_t num_config_bits
                )
{
54000f28:	7139                	addi	sp,sp,**-64**

void default_handler(void)
{
54001004:	1141                	addi	sp,sp,**-16**
54001006:	c622                	sw	s0,12(sp)

We hope this can reduce the stack space size in nested calls in our projects.

Thanks and have a nice day!

closed 2020-08-05 by hotislandn