riscv-collab/riscv-gnu-toolchain #683: How to add path in Linux?

dhruvguptadg, 2020-08-04:
I am new with linux. I am getting problem for adding path. 

I have tried export command but it shows that "not a valid identifier". How do i set the path?

Also do i need to go inside the riscv-gnu-toolchain folder and then do the "./configure --prefix = ..." and "make linux" command?

Thank you,
Dhruv Gupta

jim-wilson, 2020-08-04:
How to set the path depends on your shell.  If using a posix compatible shell, then for instance
export PATH=`pwd`:$PATH
will add the current directory to the front of the path.  You just cd into the directory you want on the path first.

If you want to build the sources, then yes, you need run the configure and make commands.  You don't have to do them inside the source tree.  You can make a build tree and run the configure command from there, for instance
mkdir build; cd build; ../riscv-gnu-toolchain/configure ...

dhruvguptadg, 2020-08-04:
Thank you i have added the path of another folder which i created in my home directory. After that i change my directory to the "riscv-gnu-toolchain" and put the "./configure --prefix=/home/Desktop/riscv-gnu-toolchain/ --with-arch=rv32i --with-abi=ilp32d", still it is giving an error that "configure: error: No acceptable C compiler found in $PATH". How to get through this problem and install the toolchain?

Thank you,
Dhruv Gupta

jim-wilson, 2020-08-04:
The README.md file has a list of prerequisites for the build.  Run the appropriate command for your flavor of linux.  Just click on code at the top of this page and scroll down to read the README.md file.

closed 2020-08-28 by dhruvguptadg