riscv-collab/riscv-gnu-toolchain #684: Problem when I use make linux, system freeze

YounesChefou, 2020-08-05:
Hello everyone,

I've been trying to build the toolchain, on a 64-bit Ubuntu 16.04 in VirtualBox, by using the make linux command. But everytime I do, I end up with the same error :
"../.././riscv-gcc/gcc/poly-int.h:1218:37: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]"

At this point, everything's frozen, the white cursor doesn't blink anymore and it's impossible to stop the process, the only thing left to do is to restart the Virtual Machine. Is the VM frozen because of the warning or is it an error not displayed in the terminal ? Should I modify the Makefile ?

Thank you for your help.

jim-wilson, 2020-08-05:
The message is just a warning and is probably unrelated to the virtualbox failure.  No riscv-gnu-toolchain source code changes are required.  I've never tried a toolchain build under virtualbox.  A toolchain build uses a lot of disk space, memory, and cpu time.  Something could be going wrong there.

YounesChefou, 2020-08-06:
Thank you, you were right ! By increasing the memory size of the VM, I managed to finish the build. Thank you again.

closed 2020-08-06 by YounesChefou