riscv-collab/riscv-gnu-toolchain #685: Cannot open linker script file riscv.ld

Shiva-prog123, 2020-08-06:
Hey ....I want to compile c code to hex in RISCV pulppissimo environment for the hello program.
I have RISCV toolchain downlaoded from
[]($ git clone https://github.com/pulp-platform/pulp-riscv-gnu-toolchain)
but the error I am getting is while using the command `riscv32-unknown-elf-gcc hello -o hello.c`

```
```
/home/sd/Desktop/RISCV/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot open linker script file riscv.ld: No such file or directory
collect2: error: ld returned 1 exit status
```

```
Showing riscv.ld directory is not found.....Is problem is with my compiler? If so is there any way to update or add riscv.ld file ?Please help me

jim-wilson, 2020-08-06:
There is no riscv.ld linker script in the GNU toolchain.  But looking at the pulp modified riscv-gnu-toolchain link you gave, I see a riscv.ld linker script in the top level dir, and a Makefile.pulp file that has a rule to install it.  So it looks like you need to do something like
   make -f Makefile.pulp build
in the build tree to install it.  or alternatively install it by hand.  Just copy the risc.ld file from the source tree to the $prefix/riscv64-unknown-elf/lib directory.

This is not a bug with riscv-gnu-toolchain.  it is a bug with pulp-riscv-gnu-toolchain.

Shiva-prog123, 2020-08-07:
Thank You Jim...

Shiva-prog123, 2020-08-07:
Hi..........
I copied the risc.ld file from the source tree to the $prefix/riscv64-unknown-elf/lib directory ....then come with another error
```
/opt/RISCV/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/lib/crt0.o: In function `_start':
(.text+0x8): undefined reference to `__global_pointer$'
collect2: error: ld returned 1 exit status
```
solved the above error by using 
`riscv32-unknown-elf-gcc -T riscv.ld add.c -o add.out -nostartfiles`

and my c code is 
```
int main() {
    int a = 4;
    int b = 12;
    while (1) {
        int c = a + b;
    }
    return 0;
}
```
I got the dump file for the above file as
`add.out:     file format elf32-littleriscv

Disassembly of section .text:

00080000 <main>:
   80000:	1101                	addi	sp,sp,-32
   80002:	ce22                	sw	s0,28(sp)
   80004:	1000                	addi	s0,sp,32
   80006:	4791                	li	a5,4
   80008:	fef42623          	sw	a5,-20(s0)
   8000c:	47b1                	li	a5,12
   8000e:	fef42423          	sw	a5,-24(s0)
   80012:	fec42703          	lw	a4,-20(s0)
   80016:	fe842783          	lw	a5,-24(s0)
   8001a:	97ba                	add	a5,a5,a4
   8001c:	fef42223          	sw	a5,-28(s0)
   80020:	bfcd                	j	80012 <main+0x12>`
another problem I am facing is some of the hex codes in the above dump file are not correct when doing manually (eg.li a5,12 actual hex code is 00c00793 but getting 47b1) is it a error?
another doubt is there any way to convert elf or dump file hex only file (means file containing only hex codes). I

jim-wilson, 2020-08-07:
The linker script should define __global_pointer$.  Looking at the pulp riscv.ld, I see a line
  _gp = . + 0x800;
the _gp should be changed to __global_pointer$.  I don't know why the pulp toolchain is broken.

0x47b1 is a compressed instruction, i.e. c.li a5,12.  The disassembler just shows it as li a5,12 as that is easier to read and most people don't care.  You can use -Mno-aliases  with objdump if you want to see the c.li instead of li for compressed instructions.

Shiva-prog123, 2020-08-08:
@jim-wilson Thanks alottt...........its working fine........I really appreciate your help. Thank you once again...........
Now my .dump file contains hex and corresponding assembly code . Another doubt is  ..is there any way to get hex code only file(.hex file ) from this .dump file?

jim-wilson, 2020-08-08:
RISC-V is little endian, so the first byte is the low order byte.  That can cause output to appear reversed in dump files depending on how the dumper works.

The readelf output should not be used as a hex file.  objcopy can convert between object file formats, and can be used to convert elf files to binary, srec, verilog, etc.  However, SiFive uses our own elf2hex program because the objcopy verilog output is buggy.
https://github.com/sifive/elf2hex

Shiva-prog123, 2020-08-09:
yes...working elf2hex and I got my hex file.........Thank you once again.......
But another error is for eg. this is my .dump line 

```
10020:	8e09         sub	a2,a2,a0
 10022:	4581         li	a1,0
```
The output hex file hex code for these two line is
`45818e09`
two lines are merging..........only for this compressed format all other working fine. I used the command `riscv32-unknown-elf-elf2hex --bit-width 32 --input test.out --output out.hex`
some part of my .dump code 
`00010008 <_start>:
   10008:	00002197          	auipc	gp,0x2
   1000c:	b2818193          	addi	gp,gp,-1240 # 11b30 <__global_pointer$>
   10010:	00001517          	auipc	a0,0x1
   10014:	33050513          	addi	a0,a0,816 # 11340 <_bss_start>
   10018:	00001617          	auipc	a2,0x1
   1001c:	34860613          	addi	a2,a2,840 # 11360 <_end>
   10020:	8e09                	sub	a2,a2,a0
   10022:	4581                	li	a1,0
   10024:	2c11                	jal	10238 <memset>`

and corresponding hex is
`00002197
b2818193
00001517
33050513
00001617
34860613
**45818e09
02972c11**
`

jim-wilson, 2020-08-09:
RISC-V has 32-bit instructions and 16-bit compressed instructions.  If you have two compressed instructions next to each other, and generate 32-bits per line in the hex dump, then yes they can end up on the same line in the hex dump.  I don't see a problem here.

If you don't want compressed instructions, then change the arch.  The default arch is rv64gc which includes the C extension for compressed instruction support.  You can use rv64g instead for instance to get code without compressed instructions.

Shiva-prog123, 2020-08-10:
Thank You for spending your valuable time............let me check with rv64gc...........................

Shiva-prog123, 2020-08-17:
Hii..... 
I loaded my hex file to the memory using readmemh command for one simple program ( addition  of two values 32(hex) + F (hex) ) result I am expecting is 41(hex) but my gtkwave simulation file loading 00000032 and 0000000F , but the result I am getting is 00102CE. I am getting the same result for adding any other values than 32 and F.
The gtkwave screenshot is attached herewith, 

[image]

Shiva-prog123, 2020-08-21:
Hi, 
My result was in Stack pointer but I didn't find any SP address in my riscv.ld linker file....where I can view stack pointer?please help me

closed 2020-09-17 by Shiva-prog123

DeekshithaMarripudi, 2023-10-30:
hi @Shiva-prog123 and @jim-wilson  where can i get the riscv.ld file.please help me

DeekshithaMarripudi, 2023-10-30:
> Hi.......... I copied the riscv.ld file from the source tree to the $prefix/riscv32-unknown-elf/lib directory ....then come with another error
> 
> ```
> /opt/RISCV/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/lib/crt0.o: In function `_start':
> (.text+0x8): undefined reference to `__global_pointer$'
> collect2: error: ld returned 1 exit status
> ```
> 
> solved the above error by using `riscv32-unknown-elf-gcc -T riscv.ld add.c -o add.out -nostartfiles`
> 
> and my c code is
> 
> ```
> int main() {
>     int a = 4;
>     int b = 12;
>     while (1) {
>         int c = a + b;
>     }
>     return 0;
> }
> ```
> 
> I got the dump file for the above file as `add.out: file format elf32-littleriscv
> 
> ```
> Disassembly of section .text:
> 
> 00080000 <main>:
>    80000:	1101                	addi	sp,sp,-32
>    80002:	ce22                	sw	s0,28(sp)
>    80004:	1000                	addi	s0,sp,32
>    80006:	4791                	li	a5,4
>    80008:	fef42623          	sw	a5,-20(s0)
>    8000c:	47b1                	li	a5,12
>    8000e:	fef42423          	sw	a5,-24(s0)
>    80012:	fec42703          	lw	a4,-20(s0)
>    80016:	fe842783          	lw	a5,-24(s0)
>    8001a:	97ba                	add	a5,a5,a4
>    8001c:	fef42223          	sw	a5,-28(s0)
>    80020:	bfcd                	j	80012 <main+0x12>`
> ```
> 
> another problem I am facing is some of the hex codes in the above dump file are not correct when doing manually (eg.li a5,12 actual hex code is 00c00793 but getting 47b1) is it a error? another doubt is there any way to convert elf or dump file hex only file (means file containing only hex codes). I

hi where u got the riscv.ld file

TommyMurphyTM1234, 2023-10-30:
> [DeekshithaMarripudi](https://github.com/DeekshithaMarripudi) commented [33 minutes ago](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/685#issuecomment-1784952562)
> hi @Shiva-prog123 and @jim-wilson where can i get the riscv.ld file.please help me

> [DeekshithaMarripudi](https://github.com/DeekshithaMarripudi) commented [21 minutes ago](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/685#issuecomment-1784970136)
> hi where u got the riscv.ld file

See here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/685#issuecomment-670077306

> [jim-wilson](https://github.com/jim-wilson) commented [on Aug 6, 2020](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/685#issuecomment-670077306)
> There is no riscv.ld linker script in the GNU toolchain. But looking at the pulp modified riscv-gnu-toolchain link you gave, I see a riscv.ld linker script in the top level dir, and a Makefile.pulp file that has a rule to install it. So it looks like you need to do ...

There is no "one size fits all" `riscv.ld` linker script  for the RISC-V bare-metal/Newlib toolchain.
The appropriate linker script depends on the target embedded/bare-metal target.

DeekshithaMarripudi, 2023-10-30:
> The linker script should define __global_pointer$. Looking at the pulp riscv.ld, I see a line _gp = . + 0x800; the _gp should be changed to __global_pointer$. I don't know why the pulp toolchain is [...]

hii jim ,i have changed _gp to_global_pointer$ but still it is showing me syntax error

TommyMurphyTM1234, 2023-10-30:
> hii jim ,i have changed _gp to_global_pointer$ but still it is showing me syntax error

@jim-wilson hasn't been that active on this forum lately so you are unlikely to get a reply from him. What you're asking about is general support with crafting linker scripts and not an issue with the `riscv-gnu-toolchain` per se. As such, it would be more appropriate to discuss the issue on another forum such as the RISC-V mailing lists, StackOverflow or a GCC user orientated forum. This Issues tracker is for specific issues/bugs with the `riscv-gnu-toolchain`.

DeekshithaMarripudi, 2023-10-30:
> @jim-wilson hasn't been that active on this forum lately so you are unlikely to get a reply from him. What you're asking about is general support with crafting linker scripts and not an issue with [...]

did u worked on riscv pulp extension?

TommyMurphyTM1234, 2023-10-30:
> did u worked on riscv pulp extension?

No. If you are trying to use Pulp then you probably should ask about these issues over there - e.g. here perhaps:

* https://github.com/pulp-platform/pulp-riscv-gnu-toolchain/issues

DeekshithaMarripudi, 2023-10-30:
i have installed pulp extension of riscv but getting erroe regarding to the linker script.
using this git repo
https://github.com/pulp-platform/pulp-riscv-gnu-toolchain

TommyMurphyTM1234, 2023-10-30:
> i have installed pulp extension of riscv but getting erroe regarding to the linker script. using this git repo https://github.com/pulp-platform/pulp-riscv-gnu-toolchain

Pulp is a completely different projed to `riscv-gnu-toolchain`.
Questions about Pulp should be directed to the Pulp project or related forums.

DeekshithaMarripudi, 2023-10-30:
kk tank you

TommyMurphyTM1234, 2023-10-30:
> hii jim ,i have changed _gp to_global_pointer$ but still it is showing me syntax error

Shouldn't that be `__global_pointer$` (double and not single undescore prefix)?

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/685#issuecomment-670676672

> [jim-wilson](https://github.com/jim-wilson) commented [on Aug 7, 2020](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/685#issuecomment-670676672)
>
> The linker script should define __global_pointer$. Looking at the pulp riscv.ld, I see a line
_gp = . + 0x800;
> the _gp should be changed to __global_pointer$. I don't know why the pulp toolchain is broken.