riscv-collab/riscv-gnu-toolchain #687: gdbserver is not building for RISCV

ghalib210, 2020-08-07:
Hi,

  I know that gdbserver still not ported to RISCV, can i know when will we get that. 
If i try to build with following step's,

$ git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
$ cd riscv-gnu-toolchain/riscv-gdb/gdb/gdbserver
$ ./configure CC=riscv64-unknown-linux-gnu-gcc --host=riscv64-unknown-linux-gnu
ERROR: 
checking for struct user_regs_struct.fs_base... no 
checking for struct user_regs_struct.gs_base... no 
checking for socklen_t... yes 
checking for Elf32_auxv_t... yes 
checking for Elf64_auxv_t... yes 
**Error: target not supported by gdbserver.** 

Regards,
Khan c.

jim-wilson, 2020-08-07:
Support was added upstream in February.  It will appear in riscv-gnu-toolchain when we next upgrade gdb.  Meanwhile you can try using the upstream sources.  This support is in the last gdb release from May.

ghalib210, 2020-08-10:
Hello Jim,

              I have tried with upstream sources also with that same issue i have found. If do have any thing can you breif them...
Thanks
Khan.c

jim-wilson, 2020-08-10:
Don't run configure in the gdbserver dir.  Run configure in the top level dir.  Don't set CC.  configure will do that if necessary.  You need --enable-gdbserver to enable gdbserver support.  You can use "make all-gdbserver" to build just gdbserver and its dependencies.  And "make insrtall-gdbserver" to install it.

You can also run configure outside the source dir.  E.g. "mkdir X-gdbserver; cd X-gdbserver; ../binutils-gdb/configure ...".  This has the advantage that you can do more than one build from the same source tree.  And cleaning up build dirs can be done with rm -rf.

ghalib210, 2020-08-11:
In Top level dir. configuration is **not recognized --enable-gdbserver Option**. As show below.

$ cd riscv-gnu-toolchain/build-gdb
$  ../configure --prefix=$PWD/install-gdb --enable-gdbserver
**configure: WARNING: unrecognized options: --enable-gdbserver**
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /bin/grep
checking for fgrep... /bin/grep -F
checking for grep that handles long lines and -e... (cached) /bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... no
checking for mpfr_init in -lmpfr... no
checking for mpc_init2 in -lmpc... no
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
**configure: WARNING: unrecognized options: --enable-gdbserver**

jim-wilson, 2020-08-12:
As mentioned previously, the gdbserver support is not in riscv-gnu-toolchain yet, but is in the FSF gdb sources.  --enable-gdbserver is a gdb configure option, not a riscv-gnu-toolchain configure option.

If gdbserver can't be built on your host, then maybe there is something wrong with your host.  Try looking at the relevant config.log files, and the relevant configure script tests, to see what the problem is.  It does build for me, natively and cross, using top of tree FSF gdb sources.

cghalibkhan, 2020-08-13:
Hello Jim,

As You said, I have build an GDBServer Successfully on **U540 Board** with FSF gdb sources as following step's,

1. git clone git://sourceware.org/git/binutils-gdb.git 
2. cd binutils-gdb/ && mkdir build install && cd build/ 
3. ../configure --prefix=$PWD/../install --enable-gdbserver 
4. make all-gdbserver 
5. make install-gdbserver 
6. ls ../install/bin/
      gdbserver

                                        **Testing:**
While Testing i have hit with an Issue like **Remote register badly formatted:**
**How to over come this Issue ?????**

**ON-TARGET MACHINE**
$ ./gdbserver localhost:2000 hello
Process /home/root/khan/riscv-binutils-gdb/install/bin/hello created; pid = 23687
Listening on port 2000
Remote debugging from host ::ffff:10.XXX.X.33, port 49940
**Remote side has terminated connection.  GDBserver will reopen the connection.**
Listening on port 2000

**ON-HOST MACHINE**
$ gdb                                                                                                                                 
..
(gdb) 
**(gdb) target remote 10.XXX.X.76:2000**
Remote debugging using 10.XXX.X.76:2000
**warning: while parsing target description (at line 4): Target description specified unknown architecture "riscv:rv64id"**
warning: Could not load XML target description; ignoring
Reading /home/root/khan/riscv-binutils-gdb/install/bin/hello from remote target...
warning: File transfers from remote targets can be slow. Use "set sysroot" to access files locally instead.
Reading /home/root/khan/riscv-binutils-gdb/install/bin/hello from remote target...
Reading symbols from target:/home/root/khan/riscv-binutils-gdb/install/bin/hello...done.
**Remote register badly formatted: T0502:40fcffff3f000000;20:005ffef73f000000;thread:p5c87.5c87;core:1;
here: 3f000000;20:005ffef73f000000;thread:p5c87.5c87;core:1;**

jim-wilson, 2020-08-13:
We don't have a dedicated gdb expert.  I suggest you try debugging it.  Or ask on one of the FSF gdb mailing lists, which you already did.

I would suggest that using rv64id could be a problem, as implied extensions are not well handled by the toolchain.  I'm not sure how you got that though.  But if you are using the gdb set arch command, then use something that is more likely to be accepted, like rv64g or rv64imafd.

rroulleau, 2021-06-02:
The gdbserver is not supported for cross compilation, what is the reason?
By enforcing the build with the cross compilation I've finally got the gdbserver executable and it run ok on a riscv target .
Got gdb (exec on a ubuntu env) to run and talk to gdbserver on the remote target (riscv).
```
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv-elf".
....
(gdb)
```
Now I am trying to get the setup to run with eclipse so I can do step by step but looking at the gdb trace it seems the gdbserver does not recognized the command "monitor" such as "monitor reset init" or  "monitor reg" send by eclipse to the target at startup time.
```
Fail to ececute MI command: monitor reset init
Fail to ececute MI command: monitor reg
```
Any clues?

jim-wilson, 2021-06-02:
gdbserver can be built cross if you configure correctly.  When you configure --host=x86_64-pc-linux-gnu --target=riscv-elf you are asking to build programs that run on x86_64 linux that produce/use riscv code.  But a RISC-V gdbserver must run on riscv not x86_64, so it can not be built this way.  You must instead configure --build=x86_64-pc-linux-gnu --host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu which produces programs that run on riscv which includes gdbserver.  This is a cross native build, and requires doing a cross build first becuase you need a cross compiler to build the gdbserver and other tools that will run on riscv.  Also, as gdbserver is a linux tool, using riscv-elf is wrong.  You must use riscv-unknown-linux-gnu or similar.  I don't think that using riscv-elf will do anything useful.  You can build just gdbserver and the the libraries it needs by using "make all-gdbserver".

As for the monitor command, I have no idea.  There are no gdb experts here.  You would have to ask someplace gdb folk hang out, like the FSF gdb mailing list.  But maybe gdbserver isn't working right if you built it wrong?  It is much easier to build it natively than via a cross native build.

rroulleau, 2021-06-07:
Thanks for the feedback. I built the gdbserver as you mentionned:
`./configure --build=x86_64-pc-linux-gnu --host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu`
Got the latest gdb distrib from binutils-gdb 11.0.50 (do not work with gdb v10.1) and eclipse 2021-03+“Mobile and Device Development” pluggins, I could run remotely step by step code on helloworld... :-)

pfaelectrique, 2021-11-19:
hello mr @jim-wilson 
i installed the riscv-gnu-toolchain ,in the source directory i have riscv-gdb but in the installation directory (bin ) i doesn't found the riscv-unknown-gdb 
i  tried to add to the configure of the toolchain enable gdb but no changes !
i can't find where is the problem or haw to force gdb to work .

riscvtest, 2021-11-19:
hello mr @jim-wilson
i installed the riscv-gnu-toolchain ,in the source directory i have riscv-gdb but in the installation directory (bin ) i doesn't found the riscv-unknown-gdb
i tried to add to the configure of the toolchain enable gdb but no changes !
i can't find where is the problem or haw to force gdb to work .

jim-wilson, 2021-11-19:
In the future, please create new issues for new problems.  Your build issue has nothing to do with gdberver.

There was probably a build failure, and you missed the error message.  This is easy to do with a parallel build.  If you have the build log file, try searching it for "error:".  If you don't have a build log, you can just run the build again  If the build succeeded, then running make again won't do anything.  If the build failed, then running make again will try to rebuild the part that failed.  You can pipe the output to a file, e.g. "make &> build.log" and then look for the error.  If you do the rebuild without a -jX option it will take longer, but it will be easier to find the error since it will be at the end.  Once you have the error, then we need to figure out why you got that error.  But without the error message it is very hard for me to guess what might be wrong.

riscvtest, 2021-11-20:
i did what u advise me to do 
in the build.log i have this :

cd build-binutils-musl && CC_FOR_TARGET=riscv64-unknown-linux-musl-gcc /home/riscv-binutils/configure \
	--target=riscv64-unknown-linux-musl \
	 \
	--prefix=/opt/riscv-musl2 \
	--with-sysroot=/opt/riscv-musl2/sysroot \
	--disable-multilib \
	 \
	--disable-werror \
	--disable-nls \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-readline

*** This configuration is not supported in the following subdirectories:
     gdbserver readline libdecnumber sim gdb
    (Any other directories should still work fine.)
*****************************************************
the gdb is disabled  even when i put  --enable-gdb it doesn't work

jim-wilson, 2021-11-20:
You are using musl.  This is an important fact.  The build output you are looking at is for the binutils build, note the cd build-binutils-musl.  So gdb is explicitly disabled here, as it isn't safe to build gdb from the binutils release branch.  There should be a cd build-gdb-musl somewhere.  Except if I look at the Makefile, I see that the people who did the musl port didn't bother to enable gdb support.  We build gdb for newlib and linux-gnu, but not for linux-musl.  So your build completed successfully, but without gdb as that isn't supported for the musl target.

If you have a riscv64-unknown-linux-gnu-gdb that will probably work.  Otherwise, if you want to build a riscv64-unknown-linux-musl-gdb, then near the top of the toplevel Makefile.in, there is this:
```
ifeq (@enable_gdb@,--enable-gdb)
newlib: stamps/build-gdb-newlib
linux: stamps/build-gdb-linux
endif
```
You need to add a line
```
musl: stamps/build-gdb-musl
```
then you need to add a rule to build stamps/build-gdb-musl.  I don't think this actually depends on glibc or musl, you can probably just copy the existing build-gdb-linux rule and change gdb-linux to gdb-musl.

I don't have musl sources handy, so I can't easily check this.  There might be some other issues that need to be fixed also.  If no one has built gdb for this target, then it isn't tested, but I would expect it to work.  If it doesn't work, then you may have to debug why gdb musl doesn't work.

riscvtest, 2021-11-21:
Thank you so much mr @jim-wilson  gdb works now

riscvtest, 2021-11-23:
i tried to debug with gdb with the  riscv64-unknown-linux-musl-gdb i have the error
Undefined target command: "sim". Try "help target".
 but it works with  riscv64-unknown-elf-gdb ,  but not with  riscv64-unknown-linux-gnu-gdb 
is the sim target is specific for the  riscv64-unknown-elf-gdb  or  there is some thong missed in the makefile ?

jim-wilson, 2021-11-23:
The gdb simulator supports embedded binaries but I don't think it will work with linux binaries.  Probably a lot of work to fix this.  Even then, it can probably only work with statically linked binaries.  If you want to try enabling it, in the gdb/configure.tgt search for the riscv rules, and note that only riscv*-*-* has a gdb_sim line, copy that to the riscv*-*-linux* stanza above.

For running linux binaries, you really should use qemu, that is the best simulator for that.  If you want better tracing than you get from qemu, then you can try using spike (riscv-isa-smi) with pk (proxy kernel).  That should work, but again, probably requires static linking.  Only qemu can run dynamically linked linux binaries.

riscvtest, 2021-11-23:
i tried to use qemu with gdb but i have problems  the  gdb does  not stop in break points that's why i want to use target sim with gdb 

 here what i did :
i run this command in the first therminal 
qemu-system-riscv64 -machine virt -m 128M -gdb tcp::1222 -S -kernel programme

in the second i run : 
test@test-VirtualBox:~/riscv_test_musl$ riscv64-unknown-linux-musl-gdb test1
GNU gdb (GDB) 10.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-linux-musl".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from test1...
(gdb) target remote :1222
Remote debugging using :1222
0x0000000000001000 in ?? ()
(gdb) load
Loading section .text, size 0xabc lma 0x10120
Loading section .rodata, size 0x11 lma 0x10be0
Loading section .eh_frame, size 0x4 lma 0x10bf4
Loading section .init_array, size 0x8 lma 0x11ff0
Loading section .fini_array, size 0x8 lma 0x11ff8
Loading section .sdata, size 0x10 lma 0x12000
Start address 0x000000000001013a, load size 2801
Transfer rate: 2735 KB/sec, 400 bytes/write.
(gdb) list main 
6	        result += i;
7	    return result;
8	}
9	
10	int main(void) 
11	{
12	    int a = sum(10);
13	    int b = sum(5);
14	    return a + b;
15	}
(gdb) b main 
Breakpoint 1 at 0x1025e: file programme.c, line 12.
(gdb) c
Continuing.

here gdb was supposed to be stopped at  the breakpoint that i have created  but in my case it remains blocked

jim-wilson, 2021-11-24:
I'm not a qemu expert, but it is suspicious that you are using qemu-system-riscv64.  This is intended for running operating systems, and can only work if you initialize the hardware first before you try to run user code.  This requires some assembly language code.  See for instance what riscv-tests does.
https://github.com/riscv-software-src/riscv-tests
https://github.com/riscv/riscv-test-env/blob/0666378f353599d01fc48562b431b1dd049faab5/p/riscv_test.h
If you try to run user code without initializing the hardware, then you will never reach main.  I tried this, and if I control-C gdb and use x/i I see that I'm executing an unimp intsruction at address 0x0 which will never make forward progress.

If you use a linux user mode qemu instead, e.g. qemu-riscv64, then a simple example should work.  It does for me.

eroom1966, 2022-04-25:
@jim-wilson 
> Support was added upstream in February. It will appear in riscv-gnu-toolchain when we next upgrade gdb. Meanwhile you can try using the upstream sources. This support is in the last gdb release from May.

I am wondering if there is any update on this ?
Thx
Lee

jim-wilson, 2022-04-25:
There has been a gdb release, but we don't do much gdb maintenance in riscv-gnu-toolchain, so we haven't upgraded yet.  That isn't one of our primary goals here.  You can always use upstream gdb sources.

TommyMurphyTM1234, 2022-09-29:
Confusing mix of unrelated or only tangentially related issues, most or all of which have been dealt with more than adequately by @jim-wilson

closed 2022-09-29 by TommyMurphyTM1234