riscv-collab/riscv-gnu-toolchain #688: which tag/branch for most recent stable release?

margej, 2020-08-07:
Hi,
Just wondering what is recommended practise to build the latest know good release? 
Is it to just checkout the tip of the release branch? I ask because this seems to be nearly 2 years old, and there are newer tags available.

jim-wilson, 2020-08-07:
The master branch, which is the default branch, is the only branch we maintain.  The tip of the master branch is the latest version.  I don't know what the release branch is for.  There are a lot of useless branches since we don't have any policy for this.

margej, 2020-08-07:
Ok, thanks. Is the tip of master bleeding edge, or likely to be relatively stable? I built master yesterday and had a number of previously unseen failures when testing (my own test built with the resulting toolchain). I saw the same errors running the binary on 2 independent RV32I targets, leading me to suspect a tool issue (or the same bug in both targets!)

Running make report-newlib also reported 15 unexpected failures. Should I be reporting these?

jim-wilson, 2020-08-07:
riscv-gnu-toolchain is maintained by volunteers who never have enough time to do all of the work that needs to be done.  riscv-gnu-toolchain includes sources for about half a dozen other independently maintained projects.  Problems with the subprojects should probably be reported to them instead of riscv-gnu-toolchain.  Note that most subprojects have large frequently changing testsuites and getting test failures down to 0 is a very large amount of work which the riscv-gnu-toolchain maintainers will never be able to do.  This kind of stuff will have to be handled in the subprojects.  Though reporting problems to the subprojects won't necessarily result in fixed, as a volunteer for one of those subprojects will have to fix it, and there is no way to predict when that will happen.  And then a volunteer for riscv-gnu-toolchain has to pull in the fixes from the subproject upstream, and there is no way to predict when that will happen.  Long term, it is better to report bugs upstream (e.g. FSF or Sourceware) rather than here, though newlib doesn't have a bug database so reporting them here or riscv- newlib might be more useful.  Reporting bugs that show up running the testsuite is probably not very useful, as anyone maintaining the sources can already see the bugs.  There just aren't enough people helping to maintain this stuff.

We do have allowlists for expected failures that we know about but haven't had time to fix yet.  Though we only have allowlists for binutils and gcc.  We haven't done this for newlib yet.  Generally, we haven't been testing newlib other than via the gcc testsuite.  An allowlist for newlib would be useful, but someone needs to volunteer to do the work.

Yes, the tip of master is meant to be stable enough for use.

margej, 2020-08-10:
Thanks, and understood. 
I'll debug the issues I'm seeing with GCC 10.1 and report upstream if it looks like a genuine riscv-gcc or riscv-binutils issue 

PS: although my toolchain build is a newlib variant, the tests I have failing are not pulling in any C library functions or startup code.

closed 2020-08-10 by margej

margej, 2020-08-14:
Just for anyone asking the same question, my issues with tip of master turned out to be due to a new output section* (.srodata.*) not being picked up by my linker script and being ommitted from the build. 

Fixing that, tip of master is working perfectly for me. Thanks again.

*or at least the presence of data in that section, where on previous versions there had been none.