riscv-collab/riscv-gnu-toolchain #689: Insanity on return value with 3 elements

fwsGonzo, 2020-08-08:
Is this a compiler bug?

```
001208c8 <my_page>:
my_page():
  1208c8:       00141737                lui     a4,0x141
  1208cc:       7e870713                addi    a4,a4,2024 # 1417e8 <__clzsi2+0x50>
  1208d0:       00e52023                sw      a4,0(a0)
  1208d4:       00144737                lui     a4,0x144
  1208d8:       22c70713                addi    a4,a4,556 # 14422c <cplus_demangle_operators+0xcb4>
  1208dc:       00e52223                sw      a4,4(a0)
  1208e0:       00100713                li      a4,1
  1208e4:       00e52423                sw      a4,8(a0)
  1208e8:       00008067                ret
```

```
struct backend_buffer
{
	const char* type;
	const char* data;
	unsigned    size;
};
#define backend_function(x) extern "C" __attribute__((used)) backend_buffer x()

backend_function(my_page)
{
	backend_buffer buffer {"1", "2", 1};
	return buffer;
}
```

It's storing to a0, which doesn't have a specific value. In my case it's zero, so it's faulting at least.

fwsGonzo, 2020-08-08:
For the record, yes it's a bug, and I solved my problem by programming ret with the right registers:

```
__attribute__((noreturn))
inline void backend_response(const char* arg0, const char* arg1, long arg2)
{
	register long a0 asm("a0") = (long) arg0;
	register long a1 asm("a1") = (long) arg1;
	register long a2 asm("a2") = arg2;

	asm volatile ("ret" : : "r"(a0), "r"(a1), "r"(a2) : "memory");
	__builtin_unreachable();
}
```

Works as expected.

sorear, 2020-08-08:
Structs larger than two pointers and not consisting entirely of floating point values are returned by writing them into a caller-allocated buffer passed in a0.  Your example is the same as:

```
void my_page(backend_buffer *__ret)
{
	backend_buffer buffer {"1", "2", 1};
	*__ret = buffer;
}
```

See https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md#integer-calling-convention .

This is a feature of most ABIs, for instance [x86_64](https://uclibc.org/docs/psABI-x86_64.pdf#page=23):

> If the type has class MEMORY, then the caller provides space for the return value and passes the address of this storage in %rdi as if it were the first argument to the function. In effect, this address becomes a “hidden” first argument. This storage must not overlap any data visible to the callee through other names than this argument

[i386](https://uclibc.org/docs/psABI-i386.pdf#page=13):

> Some fundamental types and all aggregate types are returned in memory. For functions that return a value in memory, the caller passes a pointer to the memory location where the called function must write the return value. This pointer is passed to called function as an implicit first argument. The memory location must be properly aligned according to the rules in section 2.1.1. In addition to writing the return value to the proper location, the called function is responsible for popping the implicit pointer argument off the stack and storing it in %eax prior to returning. The calling function may choose to reference the return value via %eax after the function returns.

or [aarch64](https://static.docs.arm.com/ihi0055/b/IHI0055B_aapcs64.pdf#page=19):

> Otherwise, the caller shall reserve a block of memory of sufficient size and alignment to hold the result. The address of the memory block shall be passed as an additional argument to the function in x8. The callee may modify the result memory block at any point during the execution of the subroutine (there is no requirement for the callee to preserve the value stored in x8).

fwsGonzo, 2020-08-08:
Thanks, that does indeed explain what happened. It's weird that the compiler doesn't prevent this since I clearly was not accepting a pointer to the struct as the first argument - instead I was returning it.

Either way, that explains the silliness.

closed 2020-08-08 by fwsGonzo

jim-wilson, 2020-08-08:
Structure returns use a hidden argument, similar to how the this pointer in C++ works.  There is nothing for the compiler to diagnose here.  The source code is valid, and the assembly code generated for it is correct.  You must have been calling it wrong from some assembly language code.