riscv-collab/riscv-gnu-toolchain #691: Can parallel make be invoked?

margej, 2020-08-10:
The default configuration invokes make for all submodules without the -j option 
e.g.:
`$(MAKE) -C $(notdir $@)`

Adding:

```
+NCPU:=$(shell nproc)
<snip>
$(MAKE) -j $(NCPU) -C $(notdir $@)
```

Speeds up builds significantly. Is it worth raising a PR to add this, or is it already supported by some configure switch I'm not finding?

jim-wilson, 2020-08-10:
The user is expected to add -j if desired, choosing an appropriate -j value.  Using all cpus is unfriendly on shared servers.  Sometimes doing builds with -j1 is useful to get easier to read error messages.  So there is not any one value that is correct in all cases.  In general, the GNU tools don't enable -j by default and expect the user to specify it when desired to an appropriate value.

margej, 2020-08-11:
For some reason I had convinced myself that running the top level make with -j was not working as expected, but having just tried again I see that it is. I had failed to recall that $(MAKE) invokes the sub make with the same options as make was invoked with at the top level. Clearly simply specifying -j at the top level make is the right thing to do!

closed 2020-08-11 by margej