riscv-collab/riscv-gnu-toolchain #692: How to know if current function's return address is pushed onto the stack?

TATAyu0413, 2020-08-12:
I need to know if current function's return address is pushed onto the stack. I tried to read source code but I failed to figure out how to do that. Could you give me some advice? Any help would be appreciated. Thank you in advance!

jim-wilson, 2020-08-12:
In general, the only way to determine whether the return address is in a register or on the stack is to compile with unwind info (default for C++) and then read and parse the unwind info to extract this information.  This is requires a fair amount of work to set up.

If you just need the return address, and don't need to know where it is, you can use __builtin_return_address (0).

jim-wilson, 2020-08-12:
Also note that various optimizations can confuse this, e.g. if compiling with tail call elimination, then the return address of a function might be grandparent instead of the parent, if the parent's return address was optimized away.  Whether this is a problem depends on what exactly you are doing.

TATAyu0413, 2020-08-19:
Your advice helped me a lot. Thank u so much!

closed 2020-08-19 by TATAyu0413