riscv-collab/riscv-gnu-toolchain #693: qemu riscv installation issue

YuliaM95, 2020-08-13:
Hi, I am trying to install qemu riscv using the following link -
https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html
While building linux with the command make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- defconfig
the following error shows up -
scripts/kconfig.include:35 compiler riscv64-unknown-linux-gnu-gcc not found in PATH
scripts/kconfig/Makefile:82:recipe for target 'defconfig' failed

I have included path of tool chain. Still not working. Attached the screenshot of folder structure and error.
Kindly do the needful.
[image]

sorear, 2020-08-13:
You have `/home/rashmi/riscv-gnu-toolchain/bin` in PATH, which is good, and the CROSS_COMPILE prefix also looks correct in isolation, but I'm wondering if they match each other.  Do you have `/home/rashmi/riscv-gnu-toolchain/bin/riscv64-unknown-linux-gnu-gcc` ?  If not, what do you have in that folder?

YuliaM95, 2020-08-13:
@sorear Yes, I have _riscv64-unknown-linux-gnu-gcc_ in that folder I am sure and cross-verified it as well.

sorear, 2020-08-13:
If you copy the compiler name from the make output and run it with the current PATH, is the shell able to find and run the compiler by that name?

If so, strace might help identify what's going wrong, something like `strace -f -e file make ARCH=riscv CROSS_COMPILE=riscv64-unknown-linux-gnu- defconfig` ; look for attempted accesses to gcc just before make produces the "compiler not found" error.

YuliaM95, 2020-08-13:
@sorear I am using CMake and I am using message(CMAKE_CXX_COMPILER) to print the message which compiler it is using. It is printing the same compiler. But I will try with strace now.

sorear, 2020-08-13:
I don't understand.  You''re having difficulties building Linux, which AFAIK does not use and has never used CMake.

YuliaM95, 2020-08-13:
Actually,I am trying to do profiling of the code written in C++ with the target Architecture RISC-V. The code has been cross-compiled using RISC-V GNU Toolchain. My executable is unit_tests "ELF 64-bit LSB executable, UCB RISC-V, version 1 (GNU/Linux), dynamically linked,nterpreter /lib/ld-linux-riscv64-lp64d.so.1, for GNU/Linux 4.15.0, with debug_info, not stripped" this information is retrieved using the file command.

What I am trying to do is the profiling of this using gprof. But to do the gprof gmon.out needs to be generated, to generate gmon.out the executable should be run first. I cannot run the binary elf of other architecture in some different architecture. I need a suggestion for this on which emulator or simulator does this for me or I can run on?

I have tried installing qemu using the follwing link:

https://www.google.com/url?q=https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html&sa=D&source=hangouts&ust=1597422417473000&usg=AFQjCNERr6pHYmj0SU6an3WkBRGQI52aTw

but I am getting the above errors.

jim-wilson, 2020-08-13:
It looks like there could be a typo in your PATH because it starts with a $.  If your path is OK, then "which riscv64-unknown-linux-gnu-gcc" will print the path to the binary.  If your path is broken, then that will print nothing.

If all you want is to run a binary, then you don't necessarily need linux.  You can run programs directly on qemu which is called user mode, or you can boot linux on qemu and then run programs inside linux which is called system mode.  The instructions you are following is building a system mode qemu.  You can get a user mode qemu by configuring qemu for riscv64-linux-user.  You can configure both by using e.g. "--target-list=riscv64-linux-user,riscv64-softmmu".

YuliaM95, 2020-08-13:
@jim-wilson Not just running the binary I want to the profiling of my code which has been written targetting riscv. Would be please suggest the best and fastest way to do it. I am stuck since 4 days with it, figuring out how to do profiling of my binary generated for a different architecture. I have successfully done profiling for x86 on gcc using gprof.  But now with Ubuntu I am facing many issues.

jim-wilson, 2020-08-13:
You have to run the binary to generate the profile data.  What you are doing should work.  Fix the apparent typo in your PATH environment variable.  Building a user mode qemu is an alternative solution that may be easier than the system mode qemu+linux which you are doing now.  So you could try this instead, or try both.

If you have a recent enough OS version, e.g. Ubuntu 20.04, you can just apt install packages to get a toolchain and qemu, so something like this should work
    sudo apt install qemu-user gcc-riscv64-linux-gnu-gcc
There is also a qemu-system-misc package for the riscv64 system mode qemu if you want to try running riscv64 linux.  I haven't tested it but I would expect this to work.

YuliaM95, 2020-08-15:
@jim-wilson Thank you Jim. Your suggestion really helped me to solve the issue. I followed the instructions in the mentioned link. 
[https://wiki.debian.org/RISC-V#Manual_qemu-user_installation](url)

closed 2020-08-17 by YuliaM95

dshm, 2023-11-16:
@YuliaM95  Hello! How do you get the riscv binary for profiling? using riscv64-unkonwn-elf-gcc with -pg? I am facing some question similar?

TommyMurphyTM1234, 2023-11-16:
> @YuliaM95 Hello! How do you get the riscv binary for profiling? using riscv64-unkonwn-elf-gcc with -pg? I am facing some question similar?

No need to spam the same question multiple times:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/695#issuecomment-1813931991
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/693#issuecomment-1813944894
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/551#issuecomment-1813957919

@jim-wilson explains the issue here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/771#issuecomment-727096961