riscv-collab/riscv-gnu-toolchain #694: why does gcc generate code with ebreak without any system calls being made?

HyunJe-Lee, 2020-08-14:
Hi.
I saw https://github.com/riscv/riscv-gnu-toolchain/issues/311
I checked the ebreak in the code below.
Why does gcc generate code with ebreak without any system calls being made?

gcc option: 
```
-DC_FLAGS -O2 -MMD -MP -Wall -D__NO_INLINE__ 
-mcmodel=medany -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -g
```

gcc version
```
riscv64-unknown-elf-gcc-8.3.0.exe (SiFive GCC 8.3.0-2019.08.0) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
```

C code:
```
UINT kkk = 10;
void ebreak_check_test(void)
{
        UINT stime, hwsum, celcnt;
        for (UINT testSize = 1; testSize < 8; testSize++) {
                hwsum = celcnt = 0;

                for (UINT i = 0; i < kkk; i++) {
                        stime = BenchTimeStart();
                        celcnt++;
                        hwsum += BenchTimeStop(stime);
                }

                printf("avg:%5u", hwsum / celcnt);
        }
}
```
```
0000000080015a6e <ebreak_check_test>:
{
    80015a6e:	7139                	addi	sp,sp,-64
    80015a70:	ec4e                	sd	s3,24(sp)
			hwsum += BenchTimeStop(stime);
    80015a72:	59fd                	li	s3,-1
{
    80015a74:	f04a                	sd	s2,32(sp)
    80015a76:	e852                	sd	s4,16(sp)
    80015a78:	e456                	sd	s5,8(sp)
    80015a7a:	fc06                	sd	ra,56(sp)
    80015a7c:	f822                	sd	s0,48(sp)
    80015a7e:	f426                	sd	s1,40(sp)
		hwsum = celcnt = 0;
    80015a80:	4a1d                	li	s4,7
    80015a82:	00143917          	auipc	s2,0x143
    80015a86:	c7e90913          	addi	s2,s2,-898 # 80158700 <kkk>
			hwsum += BenchTimeStop(stime);
    80015a8a:	0209d993          	srli	s3,s3,0x20
		printf("avg:%5u", hwsum / celcnt);
    80015a8e:	0012da97          	auipc	s5,0x12d
    80015a92:	d02a8a93          	addi	s5,s5,-766 # 80142790 <fivesbits+0x9260>
		for (UINT i = 0; i < kkk; i++) {
    80015a96:	00092783          	lw	a5,0(s2)
    80015a9a:	c3a1                	beqz	a5,80015ada <ebreak_check_test+0x6c>
		hwsum = celcnt = 0;
    80015a9c:	4401                	li	s0,0
    80015a9e:	4481                	li	s1,0
			stime = BenchTimeStart();
    80015aa0:	b9bea0ef          	jal	ra,8000063a <BenchTimeStart>
			hwsum += BenchTimeStop(stime);
    80015aa4:	01357533          	and	a0,a0,s3
    80015aa8:	b99ea0ef          	jal	ra,80000640 <BenchTimeStop>
		for (UINT i = 0; i < kkk; i++) {
    80015aac:	00092783          	lw	a5,0(s2)
			celcnt++;
    80015ab0:	2405                	addiw	s0,s0,1
			hwsum += BenchTimeStop(stime);
    80015ab2:	9ca9                	addw	s1,s1,a0
		for (UINT i = 0; i < kkk; i++) {
    80015ab4:	fef466e3          	bltu	s0,a5,80015aa0 <ebreak_check_test+0x32>
		printf("avg:%5u", hwsum / celcnt);
    80015ab8:	0284d5bb          	divuw	a1,s1,s0
    80015abc:	3a7d                	addiw	s4,s4,-1
    80015abe:	8556                	mv	a0,s5
    80015ac0:	001660ef          	jal	ra,8007c2c0 <_printf>
	for (UINT testSize = 1; testSize < 8; testSize++) {
    80015ac4:	fc0a19e3          	bnez	s4,80015a96 <ebreak_check_test+0x28>
}
    80015ac8:	70e2                	ld	ra,56(sp)
    80015aca:	7442                	ld	s0,48(sp)
    80015acc:	74a2                	ld	s1,40(sp)
    80015ace:	7902                	ld	s2,32(sp)
    80015ad0:	69e2                	ld	s3,24(sp)
    80015ad2:	6a42                	ld	s4,16(sp)
    80015ad4:	6aa2                	ld	s5,8(sp)
    80015ad6:	6121                	addi	sp,sp,64
    80015ad8:	8082                	ret
		printf("avg:%5u", hwsum / celcnt);
    80015ada:	9002                	ebreak
```

Thank you in advance.

aswaterman, 2020-08-14:
GCC emits trap instructions in some cases where it can detect undefined behavior, e.g., dereferencing a null pointer, or in this case, dividing by zero.

Here's a simple example that demonstrates similar behavior on a different architecture and compiler.

AArch64 MSVC: https://godbolt.org/z/cKzcEo
RV64 GCC: https://godbolt.org/z/b6a7ex

aswaterman, 2020-08-14:
Note also that `ebreak` isn't for system calls; it's a breakpoint, which is usually used to invoke the debugger.

HyunJe-Lee, 2020-08-14:
The problem does not occur with this code.

Is there any reason why ebreak doesn't happen?
Please check 'Add'.

https://godbolt.org/z/M493WY
```
UINT kkk = 10;
void ebreak_check_test(void)
{
        UINT stime, hwsum, celcnt;
        for (UINT testSize = 1; testSize < 8; testSize++) {
                hwsum = celcnt = 0;

                for (UINT i = 0; i < kkk; i++) {
                        stime = BenchTimeStart();
                        celcnt++;
                        hwsum += BenchTimeStop(stime);
                }
                if (celcnt == 0) {                                                            // Add
                	printf(".");                                                          // Add
                }                                                                             // Add
                printf("avg:%5u", hwsum / celcnt);
        }
}
```

aswaterman, 2020-08-14:
Presumably because the compiler inferred that the programmer was taking action to handle the divide-by-zero?

It is not a RISC-V-specific issue.  GCC does the same thing for AArch64 and x86-64 (the first program has a trap instruction, but the second program does not).

HyunJe-Lee, 2020-08-24:
thanks for reply.

I understood.
Thank you very much.

closed 2020-08-24 by HyunJe-Lee