riscv-collab/riscv-gnu-toolchain #695: Different ISA binary profiling results contradiction in RISC V.

YuliaM95, 2020-08-15:
I am doing profiling of my code written in CPP targeting RISC architecture. I have two binaries generated one for x86 and other for RISC-V. I have done profiling using perf and gprof. As per Theory of RISC and CISC architecture,but what I have got from perf results is contradictory. Could someone tell me what's wrong here.

Result of Perf:

Performance counter stats for './unit_tests'CISC:

180,899022      task-clock (msec)         #    0,885 CPUs utilized          
             7      context-switches                #    0,039 K/sec                  
             2      cpu-migrations                   #    0,011 K/sec                  
         1.350      page-faults                      #    0,007 M/sec                  
   588.853.057      cycles                        #    3,255 GHz                    
   863.377.707      instructions               #    1,47  insn per cycle        
   157.440.034      branches                   #  870,320 M/sec                  
       992.067      branch-misses             #    0,63% of all branches        

   0,204509183 seconds time elapsed

Performance counter stats for './unit_tests'RISC:

693,264322      task-clock (msec)         #    0,999 CPUs utilized          
            28      context-switches              #    0,040 K/sec                  
             1      cpu-migrations                  #    0,001 K/sec                  
         2.400      page-faults                    #    0,003 M/sec                  
 2.320.185.432      cycles                       #    3,347 GHz                    
 5.467.630.410      instructions              #    2,36  insn per cycle        
   960.171.812      branches                    # 1385,001 M/sec                  
     7.038.808      branch-misses             #    0,73% of all branches        

   0,693978844 seconds time elapsed

As seen from the above results the time elapsed in RISC is more than CISC and also in insn per cylce also is more in RISC. I am wondering why is it so. Can someone tell me if I am missing something or interpreting the results wrong?

sorear, 2020-08-15:
I think your immediate problem is that you're trying to use perf on qemu - you're comparing an emulation to native execution, the emulation will be much slower as a matter of course.  (Running perf _inside_ a simulation can, sometimes, give meaningful results, but qemu does not model systems at a sufficient level of detail, you would have to use a RTL simulation on a FPGA.)

That said, I would advise you to stop trying to think about the world in terms of "RISC" and "CISC".  Those terms were coined in the early 80s to describe the competition between MIPS and VAX, and they have not been relevant to practitioners in any field since the early 90s at the absolute latest.  The true CISC architectures with their rich addressing systems and 4-6 memory accesses per instruction (VAX, 68000) are thoroughly obsolete.

TommyMurphyTM1234, 2022-09-27:
Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234

dshm, 2023-11-16:
@YuliaM95  How do you do profiling for RISCV target?  using riscv64-unknown-efl-gprof? or other tools? I compiled with -pg but meet same error as shown on https://github.com/riscv-collab/riscv-gnu-toolchain/issues/771

TommyMurphyTM1234, 2023-11-16:
> @YuliaM95 How do you do profiling for RISCV target? using riscv64-unknown-efl-gprof? or other tools? I compiled with -pg but meet same error as shown on #771

@jim-wilson explains the issue here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/771#issuecomment-727096961